gopro/gpmf-parser #4: Clarification on GPS5 Altitude

garrettsickles, 2017-08-11:
The docs are ambiguous concerning the type of altitude in the GPS5 sentence. I am assuming it is height above the WGS84 ellipsoid but it could also be meters above mean sea level. I have also found your EGM 96 model in the Go Pro Quik package. It could be a useful feature for the gpmf-parser to be able to use this model to provide both height above ellipsoid and altitude (amsl).

dnewman-gpsw, 2017-08-12:
Interesting stuff, I had to read-up to make sure this wasn't mostly going over my head.  The GPS part used is the UBX-M8030 in HERO5, and we are currently using the height above the ellipsoid WGS84.  The part does offer mean sea level, but this is not currently being stored within GPS5.  I will have to ask the Quik team about the gravitational model, I wasn't aware.

garrettsickles, 2017-08-14:
Thanks for the clarification!

closed 2017-08-14 by garrettsickles