gopro/gpmf-parser #5: Syncing with video?

importon, 2017-08-16:
Is it possible to sync up a sensor’s sample precisely with the recorded pixel that happened at the same time?  I have read the readme for the parser, and I get that each sensor will have it's own timing, but I'm wondering how I would go about assigning a time to a sensor’s sample that will line up with the recorded pixel in the MP4 file.  Is this even possible? How do you guys do it over there at GoPro? Thanks for your time and all your hard work! It's very much appreciated.

allen-r-zhao, 2017-08-17:
This won't give you an exact correlation, but you can extract 1Hz GPS UTC time (from my parsing this is probably the internal clock instead of a GPS satellite cock) from the metadata. As the readme mentioned, the times are not collected at exactly 1Hz. To overcome this, you can then average the subsecond portions of each time format to figure out the approximate subsecond start time of the recording (20.40 seconds vs 20.60, for example) and interpolate from there to apply timestamps for each sample. Unfortunately, from my experiments 2 GoPros are not synced in UTC time so it's really only useful when dealing with data from one camera.

allen-r-zhao, 2017-08-17:
Additionally there's this part from the readme:

> To speed the measurement of any stream of data, the more recent bitstreams have a Key of TSMP (Total Samples) the counts the ongoing number of samples for each stream – so you don’t have to count them manually. If you compare the TSMP of the first payload with the last, you can easily determine the number of samples, then compare with the time of the video track (or MP4 clock timing), establishing the average sample rate for the GPMF data. This is not including compensation for thermal clock drifts, which do occur, although for most applications they are small enough to be safely ignored. Some streams include device temperature (GPMF Key is TMPC) for thermal clock drift or temperature sensitive calibrations.

You could use this to roughly interpolate between the starting and ending times.

importon, 2017-08-18:
The problem is the recording of the sensor's data stream does not seem to start at the same offset to when the pixels are being are being recorded in every movie file.  I'm hoping there is some sort of accurate offset value that can be obtained from the metadata for each individual movie file.

allen-r-zhao, 2017-08-18:
Ah that makes sense. Just wondering, what makes you think that the sampling offset is not consistent between videos?

importon, 2017-08-19:
When you line up the sensor data by hand with corresponding pixels it is a different amount of time that you have to offset it by in each GoPro file you've shot, even if it's the same shooting mode and frame-rate.  I'm really hoping there is something in the metadata that will tell us this unique offset for each file.

allen-r-zhao, 2017-08-21:
I'm curious as to how you lined up the data by hand, given the reasonably high sampling rates and the innate inaccuracies associated with GPS data. Did you line up with gyro/accelerometer data?

importon, 2017-08-21:
Yes, it was the imu data that was lined up with corresponding pixels by hand.  And it will be a slightly different offset for every shot even if you are shooting in the same mode and frame-rate.  I'm really hoping there is something in the metadata that will tell us this offset that is unique for each movie file.

dnewman-gpsw, 2017-08-24:
Scanline level syncing of video and IMU is very hard, particularly when they don't share a clock.  There is no metadata allows for simply extracting sub-frame temporal accuracy at this time.  IMU and video data do start up close to do each other, so frame accurate IMU metadata is less of an issue.

importon, 2017-08-24:
Thanks for the info!  How are the GoPro engineers able to sync the IMU data with the images when they do their in-camera stabilization?  A rely-able way of syncing has to exist somewhere in order for that feature to work right?

dnewman-gpsw, 2017-08-25:
Post IMU based sub-frame accurate image processing is something we considering for the future, but is not needed for in-camera stabilization.  The IMU feed live into the stabilization engine, no time stamps are not necessary.

importon, 2017-08-26:
Thanks again for the info!  Before I give up I wanted ask about one more thing.  I see that there is a  "TICK" for the steams in each payload that tells us how much time has passed in milliseconds since the camera was powered on.  Is there possibly a "TICK" from that same clock that would tell us when the video frames happened as well?

dnewman-gpsw, 2017-08-27:
TICK was for system time then the data arrived, not when it is sent, this subtle difference will make it trickier (it might be possible.)

importon, 2017-08-30:
Thanks again for the reply!  That's great news that it might still be possible.  The TICKS I've found so far are: "camera","Accelerometer","Gyroscope","GPS","Sensor gain",and "Exposure time".
Could there possibly be another one that I'm missing that tells us when the recorded image arrived in the same system time?

dnewman-gpsw, 2017-08-30:
Not any other timers, but sensor gain and exposure time are connected to the video frame timing.

importon, 2017-08-30:
Interesting. When I tried to use exposure time TICK as an indicator of how many MS to offset gyro data in order to sync with the video it did not end up helping at all, as it was very inconsistent between shots.  For example, in one shot the difference in the gyro TICK and the shutter TICK would be read 9ms, and in reality for me to line up the gyro with images by hand I would need to offset by 16ms. In another shot the difference between the gyro and shutter TICKs would be 7ms, and the amount needed to sync by hand would still be 16ms.  In yet another shot the difference between the gyro and shutter TICKs would be 8ms, and amount needed to sync by hand would be 19ms this time.  This made me think that the shutter times were not associated the time of the pixels being recorded.  Am I missing something?  Should I be able to determine sync somehow by using these shutter TICKs?  Thanks again for all your help!

dnewman-gpsw, 2017-09-03:
Any two TICK samples will have the same jitter present in any software derived clock.  However, when analysing a whole file, the jitter (noise) is reduced.  Over a longer video file you can determine the more accurate IMU clock rate as it compares to video.  However the initial offset, IMU data to video frames, might still be hard to extract beyond course frame-level accuracy.

importon, 2017-09-07:
Thank you again for the additional info!  
So when you say "hard", do you mean impossible?   Is there a possibility of some sort of math I can do to any combination of these tick values that would get me a better initial sync with IMU data and video frames?  Do you think I should give up on this initial offset sync problem?

dnewman-gpsw, 2017-09-07:
ideas on this:
1) If IMU and video frames start at exactly the same time and only the clock drift needs to be accounted for, then all the data is present to address this. This is why overlay doesn't drift out of sync.  However, these ideal start-up condition is not likely.
2) If there is some start-up delay, likely a few milliseconds, between the first IMU sample and the first video frame, there is currently no direct metadata that measures that. The TICK field in the 'SHUT' stream (shutter-speed) is related to the video timing, but it is not a direct measure of the startup conditions.  Comparing TICK in the SHUT stream with the TICK in GYRO (over time to remove jitter), might get you to sub-frame accuracy, but not near scan-line accuracy. If you would consider a doing little CV and IMU fusion you could get more accuracy. Use GYRO to find the camera's greatest change in motion, then track the timing of motion in a few frames around that point, you would get a pretty accurate video to IMU timing offset.  Using the rolling shutter "feature” to measure timing, you could deliver reasonably high precision sub-frame accuracy.

importon, 2017-09-07:
Thanks again for all your help!  The CV idea is an interesting one but it seems that it would only work in cases where CV works.  Meaning, if I were shooting in a snowstorm or scuba diving in open water there would be nothing for the CV to "see".   Does that make sense?
"If IMU and video frames start at exactly the same time and only the clock drift needs to be accounted for, then all the data is present to address this."
How would I go about testing this idea? 
Also, what do you mean by "overlay"
Thanks again!

dnewman-gpsw, 2017-09-07:
There is likely no value in syncing IMU to video, if there isn't match to see, so the CV seems approach is likely to work. To confirm 1) do a line of best fit (LoBF) for TICKs in GYRO and compare with a LoBF for TICKs in SHUT for a 5+ minute clip, there should resolve to an initial condition of the same time or with a repeatable offset. Then this would be compared with video observation-ally.  If you see a predictable offset to a denoised GYRO TICK, you can do what you want.   I haven't tried this yet.  Re:  Overlay's, I meant telemetry overlays/burn-ins.

closed 2017-09-13 by dnewman-gpsw

dnewman-gpsw, 2017-09-15:
Please experiment with v1.1 which includes a line of best fit sample rate and potential offset correction.  In the MP4reader check out // This sample code might be useful for compare data latency between channels.

importon, 2017-09-29:
Thanks for adding that.  I will give it a try.