gopro/gpmf-parser #6: UTC Time Internal or External?

allen-r-zhao, 2017-08-17:
I used a WiFi remote to start 2 GoPro Hero5 Blacks at the same time and checked to make sure they did indeed start recording simultaneously. However, when I extract their GPSU UTC timestamps, one camera thinks it's approximately 2 seconds behind the other. I checked to ensure that a fix and GPS values were obtained. Are the UTC times taken from the internal clocks or by GPS satellite broadcast? I was hoping to use UTC to sync data instead of relying on the cameras' timestamps, which can only be set by the minute.

dnewman-gpsw, 2017-08-18:
There should be in-sync with satellite extracted timing.  I will get back to this after the eclipse, I will asking internally and do some of my our experimentation.  There are other ways to sync cameras, USB connect with Quik for Desktop, the camera clock is sync'd with your PC, you can sync all your cameras this way (more than second accurate.)  You can use the same with WiFi connecting through the GoPro mobile app. The camera's internal clock will drift throughout the day, the nature of clocks, so sync just before your shoot. That all aside, GPSU should also work.

allen-r-zhao, 2017-08-18:
I did some testing with 4 GoPro Hero5 Blacks just now, all connected to GPS and a WiFi remote. Here are the UTC times I obtained. As you can see, the results are...confusing. Even if the cameras were syncing to different satellites, the difference should be miniscule. I no longer think they're based on internal clocks given how close cameras 1/3 and 2/4 are, but 2 seconds is a massive difference for GPS. 

| GOPRO 1          | GOPRO 2          | GOPRO 3          | GOPRO 4          |
|------------------|------------------|------------------|------------------|
| 170818183227.92  | 170818183229.954 | 170818183227.919 | 170818183229.955 |
| 170818183228.91  | 170818183230.944 | 170818183228.909 | 170818183230.945 |
| 170818183229.9   | 170818183231.934 | 170818183229.899 | 170818183231.88  |
| 170818183230.89  | 170818183232.924 | 170818183230.944 | 170818183232.925 |
| 170818183231.935 | 170818183233.969 | 170818183231.879 | 170818183233.915 |

dnewman-gpsw, 2017-08-19:
Thank you for that experimental data, that is truly surprising.  I will report this now to see what firmware team can tell use.  P.S.  Thank you for owning four Hero5s.

allen-r-zhao, 2017-08-19:
I appreciate it, and thank you for making this parser in the first place! Also, for anyone reading this: let me know if you'd like the (slightly changed demo) code I used to generate a .csv from the GPS data - I'm not sure if it warrants a push.

Thank my company, heh. The Hero5 is a great device for sure though; wish I could afford one at the moment.

allen-r-zhao, 2017-08-22:
I just did some further testing today and it appears my cameras are now synced to within .15 seconds of each other and .1 seconds of a random UTC clock I found on the web. Do they do weekly syncs? I'm not connecting them to any computers so it's not a computer clock sync; I'm just taking the SD card out and extracting metadata from the videos on it.

dnewman-gpsw, 2017-08-23:
Were some cameras not used much (with GPS on) before this experiment?  GPS time and UTC time differ, GPS is without leap second correction, and is currently 17 seconds ahead of UTC.  Satellite messages do occasionally update the leap seconds, so that GPS time can be corrected to UTC.  From factory conditions for the GPS receiver used, 15-seconds is the default offset (for leap seconds in 2012), but we are always looking for the correction messages, once received these are permanently stored.  Found this quote "New GPS units may not show the correct UTC time until after receiving the UTC offset message" from https://confluence.qps.nl/display/KBE/UTC+to+GPS+Time+Correction#UTCtoGPSTimeCorrection-UTCtoGPSTimeConversionTable

I repeated your findings, 2 seconds off in GPSU when I first captured after enabling GPS (one camera for the first time.)  After having GPS lock for about 10 minutes, the cameras were in UTC sync and are so after a power cycle.  I don't know how often the offset message is sent, but it didn't take too long for my cameras to be within 0.1 seconds.

allen-r-zhao, 2017-08-24:
That makes a ton of sense. While I wasn't around when the cameras were bought, I don't believe any of them had had GPS enabled before my testing. I do know that the 2 second offset persisted for an hour at the very least; I didn't measure after that.  Thanks for your help!

closed 2017-08-24 by allen-r-zhao