gopro/gpmf-parser #8: GOPRO Lat/Long seems chunky

tungalbert99, 2017-08-31:
I am getting some results from the GOPRO parser for GPS5 that makes the data look something like this:
[image]. Is there any reason that could explain this?

allen-r-zhao, 2017-08-31:
Can confirm that this occurs on multiple cameras (including the one @tungalbert99 is using), regardless of what direction the camera is moving in. If the camera is moving east-west, then the points still discretely jump from and to north-south lines; they just jump more often. We've checked the .csv file with the raw output of the demo code and they appear to match, so it might be the sensor.

KonradIT, 2017-09-02:
Seems like the GPS freq is too low. Try visualizing it using Quik for desktop.

What's the scale on that visualization by the way?

dnewman-gpsw, 2017-09-03:
What precision are you requesting the data?  If you use GPMF_ScaledData(... GPMF_TYPE_FLOAT) like shown in the demo code, some parts of the world will be limited in resolution like you are seeing.  32-bit floats are not high enough precision for world coordinates.  I should switch the sample code to double precision.  You can also read this GPS5 coordinates with GPMF_FormattedData(...) which doesn't apply the scale, presenting the number in the original 1E7 fixed-precision formatted integer.

tungalbert99, 2017-09-03:
GPS Frequency is at 18 Hz. Ok we will try using the Formatted Data command which might give us better precision. For Google Maps, this is on map view fully zoomed in so we suspected some rounding/loss of precision to be the issue. Thanks for the suggestion! We'll let you know soon how it works.

dnewman-gpsw, 2017-09-06:
I added this one to the FAQ and upgraded the demo code.

closed 2017-09-06 by dnewman-gpsw