gopro/gpmf-parser #9: Retrieve UTC timestamp from GoPro video metadata

kuku131, 2017-09-23:
Hi dnewman-gpsw,

May I know if there is away to retrieve UTC timestamp from video metadata?
Basically we are trying to achieve below feature:
When there is a large video recorded (over 4GB), it is saved as many pieces. Users will have option to copy them to Dropbox or Google Drive (file names can be changed as well in this case). 
When users import the videos into our iOS app from Dropbox, we'd like to stick them together automatically with the correct order based on the UTC timestamp. 

Many thanks!

Xiaoyue Huo

dnewman-gpsw, 2017-09-23:
Hello, 
The Hero5 Black camera will store UTC time once per second within the GPS stream, if the user had enabled GPS.  Look for the 'GPSU' FourCC tag.  More generally, all GoPro cameras store time within the MP4's timecode track (this is Quicktime standard track.)  The issue with this track, it is based on the user's time settings, with no time-zone information, so it is not always trustworthy.  However, the camera's internal clock is regularly updated with any connection to the mobile app.

David

kuku131, 2017-09-23:
Hi David,

Thanks for your prompt reply.

May I know how to extract the MP4'S timecode? Is there a library for iOS?

Regards,

Xiaoyue Huo

dnewman-gpsw, 2017-09-23:
The information is here: https://developer.apple.com/library/content/documentation/QuickTime/RM/MovieBasics/MTEditing/I-Chapter/9TimecodeMediaHandle.html

GoPro stores the timecode in frames since midnight format.   I don't know where iOS has a library for timecode support.  MP4s the consumer version of MOVs (very similar), generally MP4s don't have timecode, but as GoPro's are used professionally, there is a MOV style timecode track. The track works like all other MP4 tracks, but with much less data -- there is a single 4 byte field within the 'mdat' chuck that contains the timecode, the track tells you where it is.

kuku131, 2017-09-23:
Hi David,

Thanks very much for your support!

Regards,
Xiaoyue Huo

closed 2017-09-23 by kuku131