gopro/gpmf-parser #11: Hero 6 gyro downgrade?

importon, 2017-09-29:
Just picked up the new hero 6 black and I'm a bit shocked to that there are now 200hz worth of gyro samples per payload instead of 400hz.  Is this right?   Is the new IMU not as good as the hero 5's?

dnewman-gpsw, 2017-09-30:
Fast find, I just posted information on the new HERO6 metadata.  It is the same high quality gyro.  GYRO at 400 and ACCL at 200, still only lead to a fused IMU data stream at 200Hz -- it is the fused data that is particularly powerful for determining the camera motion.  As there is a lot of additional metadata, the GYRO rate was dropped so that payload size didn't grow too large (although there is no size limit.) 

If you have a need for higher speed GYRO data, I would like to hear that application.

closed 2017-10-01 by dnewman-gpsw

importon, 2017-10-01:
Thanks for the info.  One reason for more gyro data at the highest rate possible would be correct cmos rolling shutter distortion better.  https://www.youtube.com/watch?v=Wo4rqdynxrI

dnewman-gpsw, 2017-10-01:
On HERO6 the stabilization really good, add it uses only this gyro data.  If there wasn't good stabilization in camera, then yes more gyro data could be helpful.

importon, 2017-10-01:
Yes, however it would be great to only correct the rolling shutter and not stabilize.  This is really useful for VFX work.