gopro/gpmf-parser #12: Calculate lateral G force and Linear G force based on gmpf

kuku131, 2017-10-12:
HI David,

Hope you are well.

I have a quick question that would appreciate for your help on:
I am trying to calculate Lateral G force and Linear G force based on ACCL. We use below formula:
ACCL divided by 9.8 equals the G force.
Can you please confirm if this is correct?

Many thanks!

Xiaoyue Huo

dnewman-gpsw, 2017-10-12:
Yes, ideally that is correct.  There are small differences in the accelerometer's magnitude between cameras, you may see up to 10.2ms² for gravity on some cameras.  You can do your own camera calibration for the ACCL data if you want very precise G forces.  Simply capture data from a stationary camera with different orientations to determine the effect of gravity for each axis -- then use those numbers to compute G.

kuku131, 2017-10-13:
Hi David,

Thanks for your reply.

However the G force we calculate based on ACCL doesn't seem to be accurate at all. It is jumping everywhere on the G meter graph when we visualize it.
See video below:
https://www.dropbox.com/s/xei2wyehrde46nx/TDG-defect-G-force.mov?dl=0

May I know if there is a way to improve it?

Thanks!

Xiaoyue Huo

trackdaygenius, 2017-10-13:
Hi David, 

It's kofi - UI/UX designer from TDG team.  Our devs are having some issues, despite our calculations being correct, clearly there is something we've missed.  Compared to the G-force movement from the video telemetry overlay below , ours is spectacularly bad.  

Current GoPro visualisation: https://www.youtube.com/watch?v=wmi1GEEZSdk&t=131s

In all other circumstances, the G-force readings from TDG are fine, it's simply the way we're processing the video GPS metadata.

mvaden-gpsw, 2017-10-13:
Hi in Quik for Desktop we calculate acceleration based off of changes in GPS speed. We filter out any GPS data more than 3 standard deviations from the mean. We also filter out any GPS where we have a bad signal. Finally a rolling average is performed on on the acceleration to help smooth it.

-Mark

dnewman-gpsw, 2017-10-13:
As Mark points out, raw accelerometer data can be pretty noisy, particularly for motorsports. If you have a GPS stream, you can use that for the same purpose resulting in cleaner visualizations.  ACCL can be filtered to improve it display if you don't have a GPS5 stream.

kuku131, 2017-10-16:
Hi Mark,

Thanks for your help! May I know if it is possible to provide the core algorithm for us to smooth out the acceleration and G forces, this way we are consistent with the Quick Desktop app.

Many thanks!

Xiaoyue Huo

kuku131, 2017-10-16:
Hi Mark,

If you could share that piece of code, that would be very helpful.

Many thanks!

Xiaoyue Huo

dnewman-gpsw, 2017-10-16:
Re-opening to see what Mark can provide.

closed 2018-09-04 by kuku131