gopro/gpmf-parser #14: GPMF Format usage for write purpose

extermer, 2017-11-05:
Hi David,
I have a question regarding the format usage and possible patent issues.
Does GoPro allows the format usage only to read the GPMF data,
or is it also allowed to write data using GMPF format?

For example, record own videos with the GPMF atom inside.

BR
Georgiy

P.S.
I have read the Muvuk's issue, but my question is about legal point of view.

dnewman-gpsw, 2017-11-13:
Sorry for the delay.  We have been working on an additional open source repository for GPMF writing, so it makes it very clear of our intent to make the format open for creation as while is reading.  This should go live sometime this week.

extermer, 2017-11-14:
Brilliant!:) Thank you very much David for the answer and for a hard work developing GPMF.

closed 2017-11-14 by extermer

dnewman-gpsw, 2017-11-18:
Here is the new GPMF writer repo: https://github.com/gopro/gpmf-write