gopro/gpmf-parser #15: GoPro Hero5 Black GPS metadata

kuku131, 2017-11-14:
Hi David,

Hope all is well.

We created a video at Almeria circuit in Spain. We extracted the GPS sample data from the video file using gpmf-parser (see below).
2017-11-14 14:54:25.558 TrackDay G[32053:20237373] lat = 51.981814, lon = 34.015645
2017-11-14 14:54:25.559 TrackDay G[32053:20237373] lat = 51.981963, lon = 34.015891
2017-11-14 14:54:25.560 TrackDay G[32053:20237373] lat = 51.743817, lon = 33.234863
2017-11-14 14:54:25.560 TrackDay G[32053:20237373] lat = 51.234011, lon = 31.642675
2017-11-14 14:54:25.561 TrackDay G[32053:20237373] lat = 51.571080, lon = 32.678427
2017-11-14 14:54:25.561 TrackDay G[32053:20237373] lat = 51.576497, lon = 32.691670
2017-11-14 14:54:25.562 TrackDay G[32053:20237373] lat = 37.052133, lon = -2.214017
2017-11-14 14:54:25.562 TrackDay G[32053:20237373] lat = 37.052146, lon = -2.214036
2017-11-14 14:54:25.562 TrackDay G[32053:20237373] lat = 37.052159, lon = -2.214055
2017-11-14 14:54:25.563 TrackDay G[32053:20237373] lat = 37.052167, lon = -2.214064
2017-11-14 14:54:25.563 TrackDay G[32053:20237373] lat = 37.052172, lon = -2.214069
2017-11-14 14:54:25.563 TrackDay G[32053:20237373] lat = 37.052172, lon = -2.214067
2017-11-14 14:54:25.564 TrackDay G[32053:20237373] lat = 37.052173, lon = -2.214069
2017-11-14 14:54:25.564 TrackDay G[32053:20237373] lat = 37.052176, lon = -2.214074
2017-11-14 14:54:25.564 TrackDay G[32053:20237373] lat = 37.052177, lon = -2.214075
2017-11-14 14:54:25.565 TrackDay G[32053:20237373] lat = 37.052179, lon = -2.214084
2017-11-14 14:54:25.565 TrackDay G[32053:20237373] lat = 37.052179, lon = -2.214087
2017-11-14 14:54:25.565 TrackDay G[32053:20237373] lat = 37.052181, lon = -2.214093
2017-11-14 14:54:25.566 TrackDay G[32053:20237373] lat = 37.052185, lon = -2.214102
2017-11-14 14:54:25.566 TrackDay G[32053:20237373] lat = 37.052185, lon = -2.214105
2017-11-14 14:54:25.566 TrackDay G[32053:20237373] lat = 37.052187, lon = -2.214109
2017-11-14 14:54:25.567 TrackDay G[32053:20237373] lat = 37.052190, lon = -2.214115
2017-11-14 14:54:25.567 TrackDay G[32053:20237373] lat = 37.052193, lon = -2.214120
2017-11-14 14:54:25.568 TrackDay G[32053:20237373] lat = 37.052196, lon = -2.214127
2017-11-14 14:54:25.568 TrackDay G[32053:20237373] lat = 37.052202, lon = -2.214137
2017-11-14 14:54:25.568 TrackDay G[32053:20237373] lat = 37.052206, lon = -2.214146
2017-11-14 14:54:25.569 TrackDay G[32053:20237373] lat = 37.052212, lon = -2.214156
2017-11-14 14:54:25.569 TrackDay G[32053:20237373] lat = 37.052220, lon = -2.214170
....

Clearly the first 6 GPS coordinates are completely wrong. They are in the middle of nowhere. The rest of GPS coordinates are correct.
Can you please advise why is this happening? It seems to happen to every video.

Here is the raw video file: https://www.dropbox.com/s/jrfvb101xsqxp2b/GOPR0065.MP4?dl=0

Many thanks!

Xiaoyue Huo

dnewman-gpsw, 2017-11-14:
The first 5-6 seconds there is no GPS lock at all.  This is typically if you start capture immediately after power up (like if you use quick capture.)  After the lock begins you will get some odd positioning information, yet we provide confidence data so you can determine if you want to use it.  GPSF - GPS Fix  - 0 no lock, 2 - 2D lock, 3- full 3D lock.  In this clip we got a 2D lock at 6 seconds, and a full 3D Lock at the 24s mark.  2D lock is okay, you use GPSP to determine how accurate the lock is.  For those 6 wrong samples, the GPSF was still at 0.

kuku131, 2017-11-14:
Hi David,

Thanks for your reply. This is very helpful. It means we should start using the GPS from 2D lock.

Many thanks!

Regards,
Xiaoyue Huo

closed 2017-11-14 by dnewman-gpsw