gopro/gpmf-parser #16: Missing timestamp in GoPro metadata

kuku131, 2017-11-20:
Hi David,

Hope you had a great weekend!

I have another question regarding the GPS metadata that would appreciate for your help on:
GetGPMFSampleRateAndTimes(ms, 0.0, index, &inTime, &outTime);
Sometimes, the inTime and outTime we retrieve are empty. Can you please advise what might cause this?

Many thanks!

Regards,
Xiaoyue Huo

dnewman-gpsw, 2017-11-21:
Not really enable information here.  Please step through the code and see what fails.

kuku131, 2017-11-21:
Hi David,

Please check this video and see if you have any idea.
https://www.dropbox.com/sh/g2rjhy71np27puo/AACiMn200qLpPlKbLcWghZHma?dl=0 

Regards,
Xiaoyue Huo

dnewman-gpsw, 2017-11-21:
This capture started before GPS lock.  GPS was even reporting until the fourth seconds, and lock was established until 33 seconds into the capture.   This is all normal.  However, the mp4reader's GetGPMFSampleRate() wasn't handling the delay in the sensor start.  This fix has been pushed.

closed 2017-11-21 by dnewman-gpsw

kuku131, 2017-11-21:
Thanks David. We will try and let you know how it goes.

kuku131, 2017-11-22:
Hi David,

With the updated GPMF parser, we are able to retrieve inTime and outTime from the video above. However, we noticed that the GPS timestamp is OUT OF SYNC with the video timestamp. Can you please double check?

Many thanks!

Regards,
Xiaoyue Huo