gopro/gpmf-parser #18: Crash when parsing GPMF data from 2 videos.

kuku131, 2017-11-23:
Hi David,

We are having another issue with the latest GPMF-parser update. It always cause app to crash when parsing GPMF data from 2 video files continuously (one after another).
https://www.dropbox.com/sh/g2rjhy71np27puo/AACiMn200qLpPlKbLcWghZHma?dl=0.

Crash log:
#0. Crashed: com.apple.main-thread
0  libsystem_kernel.dylib         0x1808f411c __pthread_kill + 8
1  libsystem_pthread.dylib        0x1809c0ef8 pthread_kill + 112
2  libsystem_c.dylib              0x180865dc8 abort + 140
3  libsystem_malloc.dylib         0x180920374 szone_size + 522
4  TrackDay G                     0x1003f499c OpenGPMFSource (GPMF_mp4reader.c:424)
5  TrackDay G                     0x1001c4290 -[GPSessionParser parserWithFilePath:] (GPSessionParser.m:81)
6  TrackDay G                     0x1002a5cf4 -[TDGVideoImportViewController parserMutilSession] (TDGVideoImportViewController.m:1458)
7  TrackDay G                     0x1002a3f50 -[TDGVideoImportViewController downloadEnd] (TDGVideoImportViewController.m:1059)
8  TrackDay G                     0x10044b708 __83-[GPDownloadTask startDownloadWithMediaArray:fileName:parentDir:callback:progress:]_block_invoke (GPDownloadTask.m:77)
9  TrackDay G                     0x1004808e0 __69-[TDGRequest downloadGoProVideoWith:fileName:callback:progressBlock:]_block_invoke.221 (TDGRequest.m:355)
10 AFNetworking                   0x100b41670 __72-[AFURLSessionManagerTaskDelegate URLSession:task:didCompleteWithError:]_block_invoke_2.150 + 148
11 libdispatch.dylib              0x1807a54bc _dispatch_call_block_and_release + 24
12 libdispatch.dylib              0x1807a547c _dispatch_client_callout + 16
13 libdispatch.dylib              0x1807aab84 _dispatch_main_queue_callback_4CF + 1844
14 CoreFoundation                 0x180d10d50 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 12
15 CoreFoundation                 0x180d0ebb8 __CFRunLoopRun + 1628
16 CoreFoundation                 0x180c38c50 CFRunLoopRunSpecific + 384
17 GraphicsServices               0x182520088 GSEventRunModal + 180
18 UIKit                          0x185f26088 UIApplicationMain + 204
19 TrackDay G                     0x1000234fc main (main.m:16)
20 libdispatch.dylib              0x1807d68b8 (Missing)

The crash seems to happen here:
4  TrackDay G                     0x1003f499c OpenGPMFSource (GPMF_mp4reader.c:424)

BTW, this had never happened before the latest update push.
Can you please have a look and let me know your insight?

Many thanks!

Xiaoyue Huo

dnewman-gpsw, 2017-11-23:
The fault is in the MP4reader, which was only created for demo purposes: " *  @brief Way Too Crude MP4 reader".  I general expected most application developers would have there own.  Fortunately is was an easy fix. Now pushed.

closed 2017-11-23 by dnewman-gpsw

kuku131, 2017-11-23:
Hi David,

Thanks for this! I will try and let you know!

Regards,

Xiaoyue Huo