gopro/gpmf-parser #21: GOPRO Highlight Tags

OrbisTerrae, 2018-01-22:
Thanks for sharing the parser, super useful !

What would be the best way to extract the "highlight" tags from the GPMF file ?

Many Thanks !

dnewman-gpsw, 2018-01-22:
Hightlight tag are stored within the 'udta' atom within the header of the mp4.  It comes in two forms: the depecrated 'HMMT' atom (only stores manual highlights), or the embedded 'GPMF' atom within 'udta' that contains all the global clip data.  HERO6 uses the second type for all highlights detected. The function OpenGPMFSourceUDTA() in the demo code shows how to access this. There is DEVC with data like:

"DEVC": {
    "DVID": "HLMT",
    "DVNM": "Highlights",
    "STRM": {
      "TSMP": 3,
      "RMRK": "struct: Time (ms), in (ms), out (ms), Location XYZ (deg,deg,m), Type, Confidence (%) Score",
      "TYPE": "LLLllfFff",
      "SCAL": [1, 1, 1, 10000000, 10000000, 1, 1, 1, 1],
      "complex structure (HLMT)": "36-bytes 3 samples",
      "HLMT": [9225,9225,9225,0,0,0.00,MANL,100.00,100.00, 3865,3641,4254,0,0,0.00,FUSE,81.79,31.37, 11848,11666,11936,0,0,0.00,AIMU,57.52,28.76]
    }
  },

The TYPE is a FOURCC, some examples:
MANL - Manual - you pressed the highlight button during a capture.
AIMU - IMU based highlights -- like zero G moments.
AACD - Abrupt Audio Change Detector -- detects screams, engine starts and crash sounds. 
FUSE  - Some fused best estimate of highlights

closed 2018-01-29 by dnewman-gpsw

JuanIrache, 2019-12-05:
Hi, I'm experimenting with this and I can find the manual highlight tags in the GPMF atom, but I have never found any other TYPE (AIMU, AACD, etc.) Is this being used?