gopro/gpmf-parser #23: Improved IMU synchronization for fusion coming to other models?

importon, 2018-02-22:
I see one of the notes from the last Fusion firmware update is "Improved IMU synchronization helps with stabilization "
Is there any chance this same sort of improvement is coming to the Hero 6?  Getting the samples to line up with the video has been much harder with the Hero 6's gpmf data compared to the Hero 5 in my tests so far.  The writing of the samples into a payload at the assumed times seems to be a lot more reliable with the hero 5.  I see a lot more samples spilling over into neighboring payloads with the hero 6.  Makes it difficult to make heads or tails of how to line anything up.  Fingers crossed for an improvement on this as it's really awesome to be able to make use of this gpmf data.  Thanks again for all your hard work.  It's very much appreciated.

dnewman-gpsw, 2018-02-26:
I can't say whether the enhancement made for Fusion will be applied to Hero6 (or later models,) but I also hope so.  The need to address this was strong in Fusion as all stabilization had to be performed in post, whereas only some modes of Hero6 need the same. 
It seems you are saying there is greater sampling gitter in Hero6.  I have not looked into that.

importon, 2018-02-26:
Thanks for the reply and the additional info.  Yes, the jitter seems to be much worse in the hero 6 compared to the hero 5.  Also, the drift in the reliability of the samples being at the assumed times seems to get worse the longer the clip is.  On a long enough clip with the hero 6 you will see the timing drift gets so bad that all the samples from the later payloads are from a time associated with another payload entirely.  This doesn't happen with the Hero five as far as I can see.  The rate of drift is not consistent between videos either.   Sometimes it's too slow, sometimes it's too fast.  Do you think this is because different hardware components are used in the Hero 6 compared to the 5?  Or might this be a bug that could possible be fixed?
Thanks again for all your help.  It really is appreciated.

dnewman-gpsw, 2018-02-26:
While jitter might be greater, there should be no drift. All metadata within each payload was the data the arrived in that sampling window.  As data is not buffered, there is really no way for the data to drift, in a course sense.  Is it possible you having issues with clock extraction, to determine the sample rate of each sensor?

importon, 2018-02-27:
Thanks again for the follow up and the info.  
So here's what I'm seeing.  Maybe drift is not the right word for it.  
I've got a Hero 6 video with a payload count of 1789.  I take the total number of gyro samples, add them up, and divide by the number of payloads.  That gives me the average gyro sample rate per second.  I lay out those samples along my timeline at that rate, visually aligning them to the start of the video to within 1ms or so.  Everything looks nice and synced when you hit play.  As the clip plays for a while the samples get more and more our of sync which is expected.  I'm expecting to see things be the most out of sync around the middle of the timeline and then become more in sync by the end of the video because we're using an average rate.  However with this Hero 6 clip, when we reach the end of the clip, some of the samples from the last payload are over 1 full second out of sync with the video.  If the average rate is correct then I would expect things to get out of sync in the middle of clip but then make a recovery and become more in sync by the end of the video.  This is what I can reliably see with the hero 5.  But with the Hero 6 all bets are off. It just gets worse and worse the longer the clip is which makes me hope this is some kind of bug that can be fixed or worked around.

dnewman-gpsw, 2018-03-01:
That is a long file at near 30 minutes, is it in chapters?  I did a 22 minute capture that split into two chapters.  The GYRO for the first chapter was computed at 202.064215 Hz and had 963 payloads with a total of 194583 samples.  202.064215 x 963 = 194587.84, so it only 5 samples off -- not much jitter at all, surprisingly good.  The second chapter computed at 202.027389 with 361 payloads and 73002 samples,  361 x 202.027389 = 72931.89.  This is a bit more off, but way less that a second (0.366s.)  Why can it go off?  The computed sample rates are linear regressions, they are not averages, an average would do what you thought (off in the middle, correct at the start and end.)  The reason for the sample rate shift between first and second chapter, is the camera is heating up.  A more advance curve fit would address the thermal clock drift. Another good reason this was open sourced, need clock rate curve fitting.

importon, 2018-03-03:
Thank you so much for looking into that.  It really is super helpful and interesting to learn about this stuff. 
Yes, the 30 min clip I used was split up into 3 files.  
I think I have a better way of describing the difference I'm seeing in the Hero 6 gpmf accuracy compared to the hero 5.  
With a hero 5 clip I can get the gyro samples to line up with the video of a very long clip for it's entirety by first visually finding the offset value in ms needed to align the first payload, then when I look at the gyro tick times for all subsequent payloads, the first sample of those payloads reliably line up exactly with those tick times. 
With the hero 6 it's whole different story.  Following the same procedure, the tick times will almost never match up with the first sample of their payloads after the first couple seconds of the initially aligned clip.  That's what I mean by "samples spilling over into other payloads" and "drift".  On a longer clip, this tick+1st sample alignment can drift pretty far.  It's doesn't seem to be a linear drift either.  Sometimes it will make a recovery in the middle of the clip and then drift away again.  Sometimes not.  I can't seem to find any sort of pattern or predictability to it.  
Again I'm hoping that it's just a bug or something that can be filtered or accounted for.  Can you think of something I might be able to do to these Hero 6 gyro ticks to make them accurate like the Hero 5's?  Thanks again for all your help.

dnewman-gpsw, 2018-03-04:
Simple answer, don't use the TICK field for long during timing, that should only come from the MP4 indexing. TICK is not expected to be accurate over time, and there is no guarantee of its time base. Hardware is very different between HERO5 and HERO6, there is a possibility TICK is less accurate. TICK can be used for within the same payload and nearby payload timing (if needed), but as you can see in Fusion bitstreams, there are likely more accurate timestamps coming.

importon, 2018-03-04:
Good to know!  Thanks again.  Looking forward to seeing what comes next.

closed 2018-03-04 by importon