gopro/gpmf-parser #24: Data missing after latest firmware update

importon, 2018-03-10:
So I just tried the latest hero 6 firmware that was released a week or so ago and it looks like it no longer contains any ticks for any of the data streams?  Is this true???  Maybe they are stored in a different place?  This is a pretty big bummer.  While not perfect, this info was actually really helpfull in aligning to the video.  Why would they take this out?  Is there anything we can do to lobby for it's inclusion?  I really hope so.  This is a showstopper for all the little projects I'm working on.

dnewman-gpsw, 2018-03-10:
It is true they have been removed. If they come back, it will be in a more restricted form for which they were designed. They had limited effectiveness for stream sync'ing, as you have reported.  For sub-frame accurate sync, a new data field is required (just like on Fusion.)  The reason TICKs could not be trusted, they were pseudo timestamps, generated as the data arrived for metadata storage, not to when the data was read (a truer timestamp.)  The camera's metadata API didn't have a timestamp mechanism until that was needed in Fusion for image stabilization, the TICKs where not accurate enough.  I don't think you want TICKs back, but rather a more accurate way to determine the data phasing between video frames and IMU data.

importon, 2018-03-10:
Thanks very much for the info.  
As you say the ticks may not be ideal but they are certianly much better than nothing.  Is this new "fusion style" time info in this new firmware somewhere?  It doesn't look like I can roll back to the previous firmware either.  Do you know of a way I can roll back?  I'm kinda devastated here as I'm essentially locked out of working on my projects now.  If it's not in there now do you have any idea when this more accurate way to determine the data phasing between video frames and IMU data will make an appearance in the gpmf data?  Thanks again.  Holding my breath for some good news.

dnewman-gpsw, 2018-03-10:
You can downgrade a camera for the older firmware, although GoPro doesn't provide those downloads.  Search on ""GoPro HERO6 Black Camera Firmware"" and you will find third party links. v1.60 was the preview version.

closed 2018-03-10 by dnewman-gpsw

importon, 2018-03-10:
Thank you!  That's helpfull