gopro/gpmf-parser #27: GPMF on photo

kikislater, 2018-05-14:
Is this possible to get metadata on photos ? 
Could be interesting in science application or mapping projects to have IMU data's in exif or separate file. ...

dnewman-gpsw, 2018-05-14:
Yes this is possible.  The Exif standard doesn't have an open method for IMU data currently.  Recent GoPro cameras do use GPMF structures to store additional metadata outside of Exif.  Just a Exif uses the App 1 extension to JPEG, GPMF payload is within App 6.  However the current GoPro HERO photos do not store IMU data within App6.  The GoPro Fusion stores a limited number of IMU samples in each photo, used for horizon orientation of 360 degree images.

kikislater, 2018-05-15:
Ok what do you mean about recent Gopro ? Does GP5 black or GP5 session should able to log these IMU data's ? 
@juldebar Could be interested ...

dnewman-gpsw, 2018-05-15:
I believe this storage started with Hero6 and Fusion.  Only Fusion has IMU data

closed 2018-05-24 by dnewman-gpsw