gopro/gpmf-parser #28: Where is camera serial number located?

jin-zhe, 2018-06-28:
Hi I'm using multiple GoPro Hero 5s in my current project and I'd like to extract the camera serial number to differentiate footages from different camera. How might I extract the camera serial number from the GPMD?

dnewman-gpsw, 2018-06-28:
the 'udta' (the user data atom) contains the classic metadata ever since HERO2. It has FIRM for firmware version, CAME for camera serial/ID, LENS for lens model/vendor/batch, MUID Media Unique ID, GUMI Global Unique Media Id,  HMMT Hightlight moments.  The details of these are outside of GPMF.

closed 2018-06-28 by dnewman-gpsw

jin-zhe, 2018-06-29:
Thanks @dnewman-gpsw ! Where and how do I go about obtaining the udta? I don't see it in any of the streams