gopro/gpmf-parser #32: a heap-buffer-overflow in GPMF_parser.c:528 function GPMF_Type

Edward-L, 2018-06-30:
a heap-buffer-overflow in GPMF_parser.c:528 function GPMF_Type

```
=================================================================
==131748==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60200000efb4 at pc 0x4068ee bp 0x7ffe5153d9e0 sp 0x7ffe5153d9d8
READ of size 4 at 0x60200000efb4 thread T0
    #0 0x4068ed in GPMF_Type ../GPMF_parser.c:528
    #1 0x431a30 in PrintGPMF /opt/lxf/gpmf-parser/gpmf-parser-master/demo/GPMF_print.c:394
    #2 0x401461 in main /opt/lxf/gpmf-parser/gpmf-parser-master/demo/GPMF_demo.c:81
    #3 0x7fa688a75f44 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21f44)
    #4 0x401108 (/opt/lxf/gpmf-parser/gpmf-parser-master/demo/gpmfdemo-asan+0x401108)

0x60200000efb4 is located 0 bytes to the right of 4-byte region [0x60200000efb0,0x60200000efb4)
allocated by thread T0 here:
    #0 0x7fa688e71b0a in __interceptor_realloc (/usr/lib/x86_64-linux-gnu/libasan.so.1+0x54b0a)
    #1 0x429520 in GetGPMFPayload /opt/lxf/gpmf-parser/gpmf-parser-master/demo/GPMF_mp4reader.c:86
    #2 0x4013c8 in main /opt/lxf/gpmf-parser/gpmf-parser-master/demo/GPMF_demo.c:62
    #3 0x7fa688a75f44 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21f44)

SUMMARY: AddressSanitizer: heap-buffer-overflow ../GPMF_parser.c:528 GPMF_Type
Shadow bytes around the buggy address:
  0x0c047fff9da0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9db0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9dc0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9dd0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9de0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x0c047fff9df0: fa fa fa fa fa fa[04]fa fa fa 00 00 fa fa 00 fa
  0x0c047fff9e00: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9e10: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9e20: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9e30: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9e40: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Heap right redzone:      fb
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack partial redzone:   f4
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Contiguous container OOB:fc
  ASan internal:           fe
==131748==ABORTING
```

[poc](https://github.com/Edward-L/my-cve-list/blob/master/gpmf-parser/heap-buffer-overflow-in-GPMF_parser.c-528-GPMF_Type.poc)

dnewman-gpsw, 2018-06-30:
I just pushed some additional updates.  Please test again. Thanks.

Edward-L, 2018-06-30:
This problem has been fixed.