gopro/gpmf-parser #33: Description of tag meaning

paulmelis, 2018-07-24:
Is there some overview of what the different tags in a GPMF stream mean? Most data referenced here seems to be IMU and GPS data, but there's also some optical stuff in Fusion files it seems

dnewman-gpsw, 2018-07-25:
Fusion and more recent Hero cameras also have SHUT (shutter/exposure time) and ISOG or ISOE for sensor gain/ ISO. These are document in the readme here.  What else are you seeing that is not documented?

paulmelis, 2018-07-25:
Mostly stuff related to the lens calibrations. I can guess some of the fields, e.g. CTRX, POLY, but the exact meaning of the values isn't always clear.

dnewman-gpsw, 2018-10-04:
CTRX and POLY are part of the per-camera lens calibration, used to make stitching simpler for the real-time preview and to assist the warp stitch for high quality.  Are you working on stitching tools?  If so please ask a new question on that.

closed 2018-10-04 by dnewman-gpsw