gopro/gpmf-parser #34: Is there a way to detect how the camera is mounted?

kuku131, 2018-08-28:
Hi David,

Hope all is well.

I have a quick question that would appreciate for your help on:
Is there a way to detect how the GoPro camera is mounted? If the camera is mounted to face reverse, the 3-axis accelerometer and 3-axis gyroscope would go completely wrong. How can this be detected and handled?

Many thanks!

Xiaoyue Huo

dnewman-gpsw, 2018-08-28:
I'm not sure what you mean by IMU data being wrong.  It is always relative to the lens, so you can determine the camera orientation from the IMU data.  A mounted camera is typically under the influence of gravity, so that direction is known.

kuku131, 2018-08-29:
Hi David,

Thanks for your response!

Please see below example screen shot, my GoPro camera was mounted to face reverse. This way, the IMU.X and IMU.Y data are wrong way around. E.g: X = -X and Y = -Y. Can you please advise if there is a way to detect this?
[image]

kuku131, 2018-08-29:
I meant accel.x and accel.y are wrong way around, if the GoPro camera is mounted in the way shown in my above screen shot.
Please let me know if there is a way to detect it so that we can handle this in our app.

Thanks!

Xiaoyue Huo

dnewman-gpsw, 2018-08-30:
My understanding is this requires a sensor fusion solution.  From the IMU data you can determine a crude motion path, match that path to the GPS data and you will know the camera's orientation.

kuku131, 2018-08-30:
Thanks for the clarification David. :)

closed 2018-08-30 by kuku131