gopro/gpmf-parser #37: Trimmed GoPro videos - Is the GPS Metadata damaged?

kuku131, 2018-09-21:
Hi David,

Hope all is well.

We have some trimmed videos and would like to see if the GPS metadata can still be recognised when videos have been trimmed. Here are the trimmed videos:
https://www.dropbox.com/home/video/track/Portugal/Portimao/FE-sept2018/kofi/session4-dualCam/hack 

Many thanks!

Xiaoyue Huo

dnewman-gpsw, 2018-09-21:
That link is not available to me.  Trimming videos with GPMF requires one of these considerations:
1) You trim your in-point to match that GPMF payload (typically at 1.001Hz).  Fusion Studio 1.3 trims video with this very simple technique
2) You add an EditList to the MP4 to indicate any track to track offset.  We do this when trim camera footage and sending only a segment to the mobile app.
3) You regenerate any truncated GPMF payload.  No one has tried this yet, that I know of.  As https://github.com/gopro/gpmf-write is now public, this can be done.

closed 2018-10-03 by dnewman-gpsw

samanyousefi, 2020-05-14:
I have an issue with **gpmf-extract** module to finds the metadata track in GoPro camera when I set the video with mp4 format to **gpmfExtract** (gpmf-extract) module.

this module page on nmp :
https://www.npmjs.com/package/gpmf-extract
and also home page on git :
https://github.com/JuanIrache/gpmf-extract#readme

```
const gpmfExtract = require('gpmf-extract');
const fs = require('fs');
const path = require('path');
const util = require('util');

const dirAddress = 'E:\\mp4files';
const pathStat = util.promisify(fs.stat);

pathStat(dirAddress).then(stat => {
    if (stat) {
        if (stat.isDirectory()) {
            const folderContent = fs.readdirSync(dirAddress,'utf-8');
            if(folderContent.length){
                    folderContent.forEach(contentName => {
                        const contentSliced = contentName.split('.');
                        if(contentSliced[1] === 'mp4' || contentSliced[1] === 'MP4'){
                            const fileAddress = path.join(dirAddress,contentName);
                            gpmfExtract(fileAddress).then(res => {
                                console.log('Length of data received:', res.rawData.length);
                                console.log('Framerate of data received:', 1 / res.timing.frameDuration);
                                // Do what you want with the data
                            }).catch(e=>{
                                console.log(e,111)
                            });
                        }

                    })
            }else{
                console.log('dir is empty')
            }
        } else {
            console.log('path is not valid directory')
        }
    } else {
        console.log('path is not valid')
    }

}).catch(e => {
    console.log(e)
});

```

**this code result is :**

[0:00:00.019] [BoxParser] Unlimited box size not supported for type: ' ╝  '
[0:00:00.027] [BoxParser] Unlimited box size not supported for type: '   ╔'