gopro/gpmf-parser #38: Fusion has GPMF data?

trackdaygenius, 2018-09-24:
Hi David, 

I'm not a technical person, a lot of this stuff is beyond my domain, so forgive any seemingly obvious answers.

We've been receiving questions from our users about videos created with the FUSION camera, specifically 2 questions, whether FUSION video have GPS metadata?  I believe so, then we'd be able to extract the GPS data for lap times/telemetry and stuff, as we normally do.

Lastly, how can we allow the user to pan 360 within our app. Is there any good VR video player SDK around to use?

Any insights will be appreciated.

kofi

dnewman-gpsw, 2018-09-24:
Yes, it has pretty much of the same data as a HERO5.  However Fusion data must be stitched first and the stitched file has to preserve the GPMF metadata -  it does on mobile app, and it does on the latest version of Fusion Studio. Getting the stitched file into your ecosystem might be tricky in mobile.  360 Equirectangle to stereographic, or rectilinear projections are very easy and commonplace these days.  I will an example if you need.

closed 2018-09-24 by dnewman-gpsw

trackdaygenius, 2018-09-25:
Excellent, thanks for the insight.  If we stumble on any issues, I'll let you know

dnewman-gpsw, 2018-09-25:
Here is a demo shader I created to re-project stitched equirectangular 360 images into GoPro-like viewports.  https://www.shadertoy.com/view/XlKcWV

trackdaygenius, 2018-09-26:
> Here is a demo shader I created to re-project stitched equirectangular 360 images into GoPro-like viewports. https://www.shadertoy.com/view/XlKcWV

You legend! Thanks, I'll let my dev team review this.