gopro/gpmf-parser #39: FOV mode in Hero 5 metadata?

noobtoob4lyfe, 2018-10-03:
Hi,
First of all thanks for all the great work done here!  
I'm having trouble finding out what mode(wide,superview,narrow...) Hero 5 movies are shot in by looking at the meta data.  I see this info is in the UDTA data for Hero 6 but I don't see anything anywhere for Hero 5.  Is the shooting mode(FOV) shown anywhere in the Hero 5's metadata?

dnewman-gpsw, 2018-10-03:
HERO5 did not store global metadata in GPMF format, but this data is stored with bit flags, also within the 'udta' atom.  Look for the SETT atom for all the camera flags. Simply capture several clips in the different modes: Wide, Medium, Narrow, Linear and Superview, to see how the flags are used.

closed 2018-10-03 by dnewman-gpsw

shekh, 2018-10-04:
I'v got this piece, it may help: https://sourceforge.net/p/vdfiltermod/cch_input/ci/master/tree/gopro.cpp