gopro/gpmf-parser #51: Discrepancy between meta and video track length.

shecker, 2019-02-25:
Hi,

I want to align the video of 8 separate gopros using the GPS velocity and a signal cross-correlation. However I am noticing some 'dropped' gpmf payloads, where basically the length of the meta track (calculated using the method of timestamp generation via in/out of the GPMF_demo.c) is shorter than the video duration. 

For most cameras and their videos I'm getting a consistent length of 1061.06seconds for video and meta data. However sometimes I see a length of (1036.99s, 1045.04s, 1057.028s)  for the meta data track with a corresponding video length of (106**1**.06, 106**2**.06, 106**1**.06). Since this discrepancy affects my alignment procedure, I am wondering which duration to trust, as this also affects the solution to this issue:
1) if video and meta are, by hardware definition, always the same length, then I will need to scale my meta data.
2) if video and meta track can have different lengths, then I will need to fill in missing values for the dropped packages. 

Thanks a lot for your help.
Cheers,
Simon

dnewman-gpsw, 2019-02-25:
I'm questioning the accuracy for the metadata lengths you have provided, as they don't appear to be the standard lengths I would expect from a GoPro. However, the metadata is typically slightly shorter on all currently shipping GoPro cameras (up to one second only.)  While I personally consider this a bug, it was an intentional design decision to make sure there is no metadata weirdness over chapter boundaries.  The metadata will always line up with the IDR boundaries of the video.  IDR boundaries relate to the GOP length, and GoPro will only chapter on an IDR boundary.  So the metadata length will identically match the video length in each chapter, except the last chapter.  If you shoot 1080p50 the IDR is conveniently 1.0s exactly.  Each chapter will be around 12 minutes long (for a 45Mb/s video) so 720.0 seconds of video will have 720.0 seconds of metadata in that chapter.  If you capture 1080p50 for only 60.4 seconds of video, you get only 60.0 seconds of metadata.  If you capture 61.1 minutes, the first five 12-minute chapters will have video and metadata of matching durations, only the last minute will be missing the 0.1s of metadata from the end.  So simply, the metadata is always an IDR length (your examples where not of a known IDR length.)  The same will happen for NTSC frame rates, although the IDR length is typically 1.001s.  The MP4 correctly describes the sample frequency for each track, so there should be no mystery to the metadata and video durations.   

So you do not scale the length of the metadata to match the video, you only have metadata for the time indicated by its track in the MP4. Only the last partial second for each capture can be considered missing samples.  

You have reported something different, so either there is an error computing the track lengths correctly, or there is a camera bug.  If you dropbox link me the LRV files from those captured I can easily confirm. The LRVs and the MP4s contain identical metadata, LRV are just conveniently much smaller.

shecker, 2019-02-27:
Hi,
Thanks for the clarification.
Unfortunately I don't have the LRV files anymore, but I have uploaded 4 sequential full res. videos all shot with the same GoPro at one instance. 
I am getting a discrepancy with the meta track on the first 3 (GOPR0100.MP4, GP010100.MP4, GP020100.MP4) while the last one GP030100.MP4 seems to be fine again. 
You can find them on my website: http://people.ee.ethz.ch/~heckers/Drive360/meta_debug/ and just download via wget. Was having issues with my dropbox.
Thanks,
Simon

dnewman-gpsw, 2019-03-01:
I have confirmed your metadata durations and the camera has done something clearly wrong.  I have not seen this camera behavior before.  My own metadata analysis tool reports the length mismatch as an error:

Chapter1:
  "ERROR": "videolength is 1061.060 seconds, with 1057.056 seconds of metadata (track length)",
Chapter2:
  "ERROR": "videolength is 1062.061 seconds, with 1045.044 seconds of metadata (track length)",
Chapter3:
  "ERROR": "videolength is 1061.060 seconds, with 1037.036 seconds of metadata (track length)",
Chapter4:
  "NOTE": "videolength is 1061.060 seconds, with 1061.060 seconds of metadata",

Only the fourth chapter is correct, although the data might be out of sync due to the previous chapters.

Your firmware:
  "CAMERA FIRMWARE": "HD5.02.02.00.00",
is quite old, there are two firmware releases beyond this one (v2.5 and v2.6) for HERO5.
https://gopro.com/update/hero5

Unfortunately I don't remember, nor could I find this type of bug being mentioned before.  I don't believe a loss of GPS signal can have any impact on the metadata track duration. Currently my only speculation is the SD card was on the slower side, or an old one (they can slow with repeated use, particular the original gray-red SanDisk Ultras) as I do see system loading issues with the last few payloads.  Each payload should have around 200 samples of ACCL, yet in chapter 1, the second to last payload has 241 samples and the last has 399.  This suggests the camera was having trouble creating the next capture and wasn't able to service the metadata in a timely manner (a lower priority than the video.)  This shows in the beginning of chapter 2, which has many of the early payloads with too many samples as the system was trying to catch up.  If the camera does take too long to service the metadata, buffers can be exceeded and samples are lost. I haven't confirmed that has happened here, as it seems that all the data is there but time-stamped incorrectly.  In this case use the video duration, if the video duration is an IDR length -- which it is for chaptered video.

closed 2019-03-01 by dnewman-gpsw