gopro/gpmf-parser #52: 25fps bug?

noobtoob4lyfe, 2019-02-28:
I noticed on the Hero 7 that if you takes the TSMP of the gyro stream and divide it by the number of payloads it will give you a number around 200 which is the spec of the gyro recording rate from the readme.  But if you're shooting in 25fps and do the same TSMP/payloads, the number ends up being way off from 200.  Is this a bug?  Is there a better way to figure out the average rate per payload for 25fps footage?  Thanks!

dnewman-gpsw, 2019-03-01:
This is not a bug.  25fps might seem like a special case (it is not if you read the MP4 header correctly) as the IDR length is 26 frames not 25.  This means the duration of the payload is 26/25 or 1.04s. On average you will have 208 ACCL and GYRO samples per payload, so it is still around 200Hz.  50/100fps modes have a payload duration of 1.0s. All the NTSC frame rates are 1.001s per payload.  You don't need to hard-code any of this as the MP4 track for metadata does indicate the correct duration for each payload for all GoPro modes.

closed 2019-03-01 by dnewman-gpsw

noobtoob4lyfe, 2019-03-03:
That's very helpful thanks!