gopro/gpmf-parser #55: New 4 letter codes

dchannin, 2019-04-20:
Yhank you for providing this repo. Can you provide doc for:
FWVS
KBAT
GPRI
ATTD
GLPI
VFRH
SYST
BPOS
ATTR
SIMU
ESCS
SCPR
LNED
CYTS
CSEN

dnewman-gpsw, 2019-04-21:
These seem to be Karma drone GPMF streams. If you look in the sample bitstream it is mostly self documenting.  e.g KBAT is preceded with STNM (stream name) of "Battery Status", and a UNIT that describes many of the fields.  Karma used MAVLink status flight control and stored that wrapped as GPMF.  A nice GPMF benefit is that any existing C structures can be described is a complex data type in GPMF.  I think you will discover that many of these have similar MAVlink messages https://mavlink.io/en/messages/common.html    Consider LNED, with name "Local Position NED", I don't know what that is.  It has UNIT as ["s", "m", "m", "m", "m/s", "m/s", "m/s"], with a struct TYPE: "Lffffff", in the MAVLink docs that is very similar to LOCAL_POSITION_NED_COV  

time_usec | uint64_t | us |  
x | float | m |   | X Position
y | float | m |   | Y Position
z | float | m |   | Z Position
vx | float | m/s |   | X Speed
vy | float | m/s |   | Y Speed
vz | float | m/s |   | Z Speed

So while I don't have documentation, most should be straight forward to work out.

dchannin, 2019-04-21:
Thank you very much!
Dave

closed 2019-04-21 by dnewman-gpsw