gopro/gpmf-parser #56: Use of MTRX

dchannin, 2019-04-20:
Hello,

Am I correct that when MTRX is present, it should be used as a 3x3 matrix to re-orient the ACCL data?

Thank you,
Dave

dnewman-gpsw, 2019-04-20:
Yes, optionally.  The plan is to have the GPMF_ScaledData() call apply all the SCAL and MTRX manipulations automatically (only only does scale currently.)  MTRX, ORIN and ORIO all can be used.  MRTX might eventually carry the calibration for the orientation, that is why it is an array of floats, but it doesn't today.  ORIN - describes the stored (input) orientation, ORIO is the recommended output orientation (which will match HERO5 for consistency.)  In HERO7 we have:
      "ORIN": "YxZ",
      "ORIO": "ZXY",

lower case indicates a sign reversal.
The above means:
      outACCL[0] = inACCL[2];
      outACCL[1] = -inACCL[1];
      outACCL[2] = inACCL[0];

dchannin, 2019-04-20:
Thank you. I am running my own Python code so I do the SCAL and MTRX myself.

closed 2019-04-20 by dchannin