gopro/gpmf-parser #58: Relation between frames and metadata

JuanIrache, 2019-05-16:
Hi David,

Thanks for publishing this documentation. It's great.

I am working on a [JavaScript parser](https://github.com/JuanIrache/gopro-telemetry), and while I can access all the metadata successfully I am a bit lost as to where the time information should come from.

Sorry if this has been asked before, but I could not find a clear explanation (and have no experience in C, so deducing it from the repo is not easy for me).

I can read and intrpolate the GPSU values, but that does not always work (if the user did not enable GPS, or on the Hero5 Session). I guess I have to use the timecode track of the MP4, but how does that relate to the DEVC packets timing? I don't necessarily need UTC timestamps. Frames or milliseconds since the recording started would do too.

Could you (or someone) point me in the right direction?

Much appreciated.

dnewman-gpsw, 2019-05-16:
Timing is extracted from the MP4 indexing.  Each MP4 track has a presentation time and duration for each payload, same is true for GPMF data in the MP4.  This is why it is not extensively documented here, it is using standard mechanisms.  However, where it differs from MP4 is documented in the GPMF Timing and Clocks segment of the readme.  Time of day is not always stored in GPMF (GPS has to be on, yes.)

JuanIrache, 2019-05-16:
Thank you, that's helpful

closed 2019-05-16 by JuanIrache