gopro/gpmf-parser #60: Multiple crashes when parsing MP4 files

HongxuChen, 2019-05-17:
As of e27b702, when running `./gpmf-parser $FILE`, it may report with crashes when building with AddressSanitizer.

Please see below for details.

HongxuChen, 2019-05-17:
```
==4499==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x602000000030 at pc 0x000000551cf6 bp 0x7ffeec007e90 sp 0x7ffeec007e88
WRITE of size 8 at 0x602000000030 thread T0
    #0 0x551cf5 in OpenMP4Source /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_mp4reader.c:394:32
    #1 0x549106 in main /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_demo.c:48:15
    #2 0x7f565af92b96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #3 0x41c209 in _start (/home/exp/FOT/gpmf/gpmf-asan/BUILD/gpmf-parser+0x41c209)

0x602000000031 is located 0 bytes to the right of 1-byte region [0x602000000030,0x602000000031)
allocated by thread T0 here:
    #0 0x4d1860 in malloc (/home/exp/FOT/gpmf/gpmf-asan/BUILD/gpmf-parser+0x4d1860)
    #1 0x5516c2 in OpenMP4Source /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_mp4reader.c:376:41
    #2 0x549106 in main /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_demo.c:48:15
    #3 0x7f565af92b96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_mp4reader.c:394:32 in OpenMP4Source
Shadow bytes around the buggy address:
  0x0c047fff7fb0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c047fff7fc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c047fff7fd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c047fff7fe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c047fff7ff0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c047fff8000: fa fa 00 04 fa fa[01]fa fa fa fa fa fa fa fa fa
  0x0c047fff8010: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff8020: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff8030: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff8040: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff8050: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==4499==ABORTING
```

[hbo_GPMF_mp4reader.c:394_1.mp4.zip](https://github.com/gopro/gpmf-parser/files/3190795/hbo_GPMF_mp4reader.c.394_1.mp4.zip)

HongxuChen, 2019-05-17:
```
==4546==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6220000436b4 at pc 0x00000050f60a bp 0x7ffce32da850 sp 0x7ffce32da848
READ of size 4 at 0x6220000436b4 thread T0
    #0 0x50f609 in GPMF_Next /home/exp/FOT/gpmf/gpmf-asan/GPMF_parser.c:300:11
    #1 0x510488 in GPMF_FindNext /home/exp/FOT/gpmf/gpmf-asan/GPMF_parser.c:344:16
    #2 0x558c3a in GetGPMFSampleRate /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_mp4reader.c:774:39
    #3 0x549d03 in main /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_demo.c:242:19
    #4 0x7f1d50a2cb96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #5 0x41c209 in _start (/home/exp/FOT/gpmf/gpmf-asan/BUILD/gpmf-parser+0x41c209)

0x6220000436b4 is located 0 bytes to the right of 5556-byte region [0x622000042100,0x6220000436b4)
allocated by thread T0 here:
    #0 0x4d1c85 in realloc (/home/exp/FOT/gpmf/gpmf-asan/BUILD/gpmf-parser+0x4d1c85)
    #1 0x54a0a6 in GetPayload /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_mp4reader.c:62:27
    #2 0x558b9e in GetGPMFSampleRate /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_mp4reader.c:762:22
    #3 0x549d03 in main /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_demo.c:242:19
    #4 0x7f1d50a2cb96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/exp/FOT/gpmf/gpmf-asan/GPMF_parser.c:300:11 in GPMF_Next
Shadow bytes around the buggy address:
  0x0c4480000680: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4480000690: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c44800006a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c44800006b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c44800006c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c44800006d0: 00 00 00 00 00 00[04]fa fa fa fa fa fa fa fa fa
  0x0c44800006e0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c44800006f0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4480000700: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4480000710: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4480000720: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==4546==ABORTING
```
[hbo_GPMF_parser.c:300_1.mp4.zip](https://github.com/gopro/gpmf-parser/files/3190797/hbo_GPMF_parser.c.300_1.mp4.zip)
[hbo_GPMF_parser.c:300_2.mp4.zip](https://github.com/gopro/gpmf-parser/files/3190798/hbo_GPMF_parser.c.300_2.mp4.zip)

HongxuChen, 2019-05-17:
```
ASAN:DEADLYSIGNAL
=================================================================
==4642==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x00000054a09a bp 0x7ffea718e230 sp 0x7ffea718e160 T0)
==4642==The signal is caused by a READ memory access.
==4642==Hint: address points to the zero page.
    #0 0x54a099 in GetPayload /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_mp4reader.c:62:56
    #1 0x54936f in main /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_demo.c:94:14
    #2 0x7f5b84f40b96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #3 0x41c209 in _start (/home/exp/FOT/gpmf/gpmf-asan/BUILD/gpmf-parser+0x41c209)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_mp4reader.c:62:56 in GetPayload
==4642==ABORTING
```

[npr_r_GPMF_mp4reader.c:62_1.mp4.zip](https://github.com/gopro/gpmf-parser/files/3190806/npr_r_GPMF_mp4reader.c.62_1.mp4.zip)
[npr_r_GPMF_mp4reader.c:62_2.mp4.zip](https://github.com/gopro/gpmf-parser/files/3190807/npr_r_GPMF_mp4reader.c.62_2.mp4.zip)

HongxuChen, 2019-05-17:
```
ASAN:DEADLYSIGNAL
=================================================================
==4715==ERROR: AddressSanitizer: SEGV on unknown address 0x60240000002c (pc 0x0000005508b8 bp 0x7ffff3993d90 sp 0x7ffff3992620 T0)
==4715==The signal is caused by a READ memory access.
    #0 0x5508b7 in OpenMP4Source /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_mp4reader.c:342:34
    #1 0x549106 in main /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_demo.c:48:15
    #2 0x7fc19afeab96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #3 0x41c209 in _start (/home/exp/FOT/gpmf/gpmf-asan/BUILD/gpmf-parser+0x41c209)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_mp4reader.c:342:34 in OpenMP4Source
==4715==ABORTING
```

[r_GPMF_mp4reader.c:342_1.mp4.zip](https://github.com/gopro/gpmf-parser/files/3190810/r_GPMF_mp4reader.c.342_1.mp4.zip)

HongxuChen, 2019-05-17:
```
ASAN:DEADLYSIGNAL
=================================================================
==4765==ERROR: AddressSanitizer: SEGV on unknown address 0x7f3da3bad5c8 (pc 0x00000050d821 bp 0x7ffeeced2c90 sp 0x7ffeeced25c0 T0)
==4765==The signal is caused by a READ memory access.
    #0 0x50d820 in GPMF_Next /home/exp/FOT/gpmf/gpmf-asan/GPMF_parser.c:224:25
    #1 0x510488 in GPMF_FindNext /home/exp/FOT/gpmf/gpmf-asan/GPMF_parser.c:344:16
    #2 0x549cba in main /home/exp/FOT/gpmf/gpmf-asan/demo/GPMF_demo.c:237:21
    #3 0x7f3da29d0b96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #4 0x41c209 in _start (/home/exp/FOT/gpmf/gpmf-asan/BUILD/gpmf-parser+0x41c209)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/exp/FOT/gpmf/gpmf-asan/GPMF_parser.c:224:25 in GPMF_Next
==4765==ABORTING
```

[r_GPMF_parser.c:224_1.mp4.zip](https://github.com/gopro/gpmf-parser/files/3190815/r_GPMF_parser.c.224_1.mp4.zip)
[r_GPMF_parser.c:224_2.mp4.zip](https://github.com/gopro/gpmf-parser/files/3190816/r_GPMF_parser.c.224_2.mp4.zip)

HongxuChen, 2019-05-18:
```
ASAN:DEADLYSIGNAL
=================================================================
==30501==ERROR: AddressSanitizer: SEGV on unknown address 0x602c00000004 (pc 0x00000054f145 bp 0x7ffdff46ccf0 sp 0x7ffdff46b580 T0)
==30501==The signal is caused by a READ memory access.
    #0 0x54f144 in OpenMP4Source /home/exp/FOT/gpmf/gpmf-parser/demo/GPMF_mp4reader.c:298:42
    #1 0x549106 in main /home/exp/FOT/gpmf/gpmf-parser/demo/GPMF_demo.c:48:15
    #2 0x7fadf0e89b96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #3 0x41c209 in _start (/home/exp/FOT/gpmf/gpmf-parser/BUILD/gpmf-parser+0x41c209)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/exp/FOT/gpmf/gpmf-parser/demo/GPMF_mp4reader.c:298:42 in OpenMP4Source
==30501==ABORTING
```
[r_GPMF_mp4reader.c:298_1.mp4.zip](https://github.com/gopro/gpmf-parser/files/3193803/r_GPMF_mp4reader.c.298_1.mp4.zip)

HongxuChen, 2019-05-18:
```
ASAN:DEADLYSIGNAL
=================================================================
==30387==ERROR: AddressSanitizer: SEGV on unknown address 0x60240000004c (pc 0x0000005528db bp 0x7ffc1ec10730 sp 0x7ffc1ec0efc0 T0)
==30387==The signal is caused by a READ memory access.
    #0 0x5528da in OpenMP4Source /home/exp/FOT/gpmf/gpmf-parser/demo/GPMF_mp4reader.c:447:37
    #1 0x549106 in main /home/exp/FOT/gpmf/gpmf-parser/demo/GPMF_demo.c:48:15
    #2 0x7f6ac2e70b96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #3 0x41c209 in _start (/home/exp/FOT/gpmf/gpmf-parser/BUILD/gpmf-parser+0x41c209)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/exp/FOT/gpmf/gpmf-parser/demo/GPMF_mp4reader.c:447:37 in OpenMP4Source
==30387==ABORTING
```

[r_GPMF_mp4reader.c:447_1.mp4.zip](https://github.com/gopro/gpmf-parser/files/3193805/r_GPMF_mp4reader.c.447_1.mp4.zip)

HongxuChen, 2019-05-18:
```
ASAN:DEADLYSIGNAL
=================================================================
==30556==ERROR: AddressSanitizer: SEGV on unknown address 0x60240000002c (pc 0x000000550de8 bp 0x7ffeabd9fc10 sp 0x7ffeabd9e4a0 T0)
==30556==The signal is caused by a WRITE memory access.
    #0 0x550de7 in OpenMP4Source /home/exp/FOT/gpmf/gpmf-parser/demo/GPMF_mp4reader.c:351:32
    #1 0x549106 in main /home/exp/FOT/gpmf/gpmf-parser/demo/GPMF_demo.c:48:15
    #2 0x7fee01ff3b96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #3 0x41c209 in _start (/home/exp/FOT/gpmf/gpmf-parser/BUILD/gpmf-parser+0x41c209)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/exp/FOT/gpmf/gpmf-parser/demo/GPMF_mp4reader.c:351:32 in OpenMP4Source
==30556==ABORTING
```

[w_GPMF_mp4reader.c:351_1.mp4.zip](https://github.com/gopro/gpmf-parser/files/3193806/w_GPMF_mp4reader.c.351_1.mp4.zip)

dnewman-gpsw, 2019-05-22:
I will check this out soon.

dnewman-gpsw, 2019-05-28:
All this should now be fixed on both develop and master branches.  The all the errors seems to be in two categories, reading past the end of the file or allocating 0 bytes for offsets and/or frame sizes.

closed 2019-05-28 by dnewman-gpsw

HongxuChen, 2019-06-05:
Hi @dnewman-gpsw , with 50d4198, it seems that it still crashes on `hbo_GPMF_parser.c:300_1.mp4` (see above).

Another POC (zipped) and its ASAN output:
[hbo_GPMF_parser.c:300_3.mp4.zip](https://github.com/gopro/gpmf-parser/files/3256899/hbo_GPMF_parser.c.300_3.mp4.zip)
[hbo_GPMF_parser.c:300_3.asan.txt](https://github.com/gopro/gpmf-parser/files/3256898/hbo_GPMF_parser.c.300_3.asan.txt)

dnewman-gpsw, 2019-06-05:
While I couldn't reproduce this failure, I did close a few potential points of failure based on the asan.txt output.

dnewman-gpsw, 2019-06-05:
Thank you for all these investigations.

ShangzhiXu, 2025-03-20:
CVE-2019-15147 for https://github.com/gopro/gpmf-parser/issues/60#issuecomment-493387848 
CVE-2019-15146 for https://github.com/gopro/gpmf-parser/issues/60#issuecomment-493386950
CVE-2019-15148 for https://github.com/gopro/gpmf-parser/issues/60#issuecomment-493657698