gopro/gpmf-parser #63: How to interpret TICK

JuanIrache, 2019-06-07:
Hello David,

I am not sure how to interpret the TICK key. According to the documentation, it represents the beginning of the processing. So given this simplified structure (hero6.mp4 sample from this repo):

> DEVC
>    DVNM	:	Hero6 Black
>    TICK	:	339463
>    STRM
>       TICK		:	339474
>       STNM	:	Gyroscope (z,x,y)
>       GYRO
>          samples...

Am I right in assuming that the difference between the STRM TICK and the DEVC TICK will give me the offset to find where the sample happened in the mp4 time? In this case 11 milliseconds after the payload time.

dnewman-gpsw, 2019-06-07:
You can use them that way, but they were discontinued for being a misleading crutch, as they are less accurate than the timing techniques demo'd in GetGPMFSampleRate().  The TICK fields were removed from v2 firmware of HERO6 and are not in HERO7.  The accuracy issue is that timestamps, when software based (like these) are noisy from the OS's task switching.  While 339474-339463 (11ms) could be the time difference for the first GYRO sample, or it could be that a system interrupt delayed the storage of the GYRO data.

closed 2019-06-07 by dnewman-gpsw

JuanIrache, 2019-06-07:
Appreciated

noobtoob4lyfe, 2019-06-28:
Hi David.

I just wanted to chime in and give my vote for the return of the ticks to futures firmwares.  We made a video stabilization app that uses the GPMF gyro data, and we found the ticks to be incredibly useful for syncing the gyro data with the video data.  You can see the process in action here https://vimeo.com/328333594 . 
Without the ticks, we found that we needed to adjust and re-sync things every 15 second in order for it to look decent.  With ticks, we found that we only needed 1 or 2 syncs for the entire clip no matter how long.  Did we discover a use for the ticks that wasn't originally intended?  Is there a way to get better sync without ticks that we're missing?
Thanks again for your help!

dnewman-gpsw, 2019-07-05:
Sorry for the delay, I've been on vacation. 

Your sync approach is valid with existing HERO cameras.  While you are using the TICK field in the way we had imagined, we removed them as they were not sufficient for fully automatic stabilization (metadata only), as you have found in requiring video analysis for sync.  In Fusion we have far more precision than TICK, as the post stabilization includes sub-frame (up to 1/100th of a frame) rolling shutter correction. Fusion has a much slower read out speed than HERO5/6/7, so the rolling shutter would be significant if is wasn't for the metadata precision.  The new metadata is STMP, for timeSTaMP, this the μs time for when the data was sent, whereas TICK was the ms time for when the data arrived for storage (much more jitter in arrival time.)  In Fusion we also add STMP for video originated metadata like SHUT, so the timestamp difference between SHUT and GYRO is the metadata alignment for sync.

Going forward we hope to improve the precision on new HERO cameras to me more like Fusion, along with other metadata improvements.

dnewman-gpsw, 2019-07-05:
P.S. ReelSteady GO is a much appreciated and awesome outcome of open sourcing GPMF.  Please add a option to preserve some if not all fisheye curvature. My reasoning: https://cineform.blogspot.com/2016/04/lens-curvature-and-embracing-its.html

noobtoob4lyfe, 2019-08-06:
Sorry for the delay.
That's great!  Here's hoping the Hero 8 includes STMP in the metadata.  That's also an interesting write-up about lens distortion.  We'll see what we can do about a tool to reintroduce some distortion.  
Thanks again!

dnewman-gpsw, 2019-08-07:
And more recently I posted the math for a stock GoPro lens, forward and reverse polynomials. https://cineform.blogspot.com/2019/07/gopro-virtual-lens.html