gopro/gpmf-parser #68: Truncated FACE data?

JuanIrache, 2019-08-12:
Hi,

In some videos shot with the latest Hero7 Black firmware I'm getting data that seems truncated, specifically in the FACE stream. I get the STNM, TYPE and TIMO fourCCs, but not FACE. This is a simplified output snippet:

```
{
      "DVID": 1,
      "DVNM": "Hero7 Black",
      "STRM": [
        {
          "STNM": "Face Coordinates and details",
          "TYPE": "Lffffffffffffffffffffff",
          "TIMO": [0.7540000081062317]
        },
        { "STNM": "Sensor ISO", "ISOE": [1593] },
        { "STNM": "Exposure time (shutter speed)", "SIUN": "s", "SHUT": [0.03000163473188877] }
      ]
    }
```
In other packets, though, the FACE fourCC is there even if it just contains [null]. I'm wondering if I'm interpreting something wrong (I use my own JavaScript parser). Here's the raw data in case it helps: https://tailorandwayne.com/GH010077.bin

Thank you

dnewman-gpsw, 2019-08-12:
The camera looks like it was shooting in Time-Lapse Video (TLV) mode. TLV is a peculiar case, as the GPMF is down sampled.  Looks like a minor camera bug, the downsampling should have left an empty FACE, if no faces are detected.  In my parser in see the same behavior.

JuanIrache, 2019-08-12:
Thanks. Indeed, it happens with time-lapses.

closed 2019-08-12 by JuanIrache