gopro/gpmf-parser #74: heap overflow in GPMF_Next  GPMF_parser.c:300

cuanduo, 2019-10-15:
Tested in Ubuntu 19.04, 64bit, master(ceb3815)
compiled by gcc -g -fsanitize=address
Triggered by
$ gpmf-parse $POC

POC file
[poc0.zip](https://github.com/gopro/gpmf-parser/files/3727762/poc0.zip)
```
root@ubuntu:/home/tim/gpmf-parser-asan/build# ./gpmf-parser ../crashes/karma.mp4-out_of_bound-idx\:0x24-0x0 
  STRM of ACCL of type s with 196 samples -- 3 elements per sample
  STRM of GYRO of type s with 392 samples -- 3 elements per sample
  STRM of ISOG of type f with 29 samples 
  STRM of SHUT of type f with 29 samples 
  STRM of FWVS of type c with 8 samples 
  STRM of KBAT of type lLlsSSSSSSSBBBb with 1 sample -- 15 elements per sample
  STRM of GPRI of type JlllSSSSBB with 4 samples -- 10 elements per sample
  STRM of ATTD of type LffffffB with 4 samples -- 8 elements per sample
  STRM of GLPI of type LllllsssS with 4 samples -- 9 elements per sample
  STRM of VFRH of type ffffsS with 4 samples -- 6 elements per sample
  STRM of SYST of type JJ with 1 sample -- 2 elements per sample
  STRM of BPOS of type lllfffff with 2 samples -- 8 elements per sample
  STRM of ATTR of type Jffff with 4 samples -- 5 elements per sample
  STRM of SIMU of type Lsssssssss with 4 samples -- 10 elements per sample
  STRM of ESCS of type JSSSSSSSSssssSSSSSSSSSSSSSSSSB with 1 sample -- 30 elements per sample
  STRM of LNED of type Lffffff with 4 samples -- 7 elements per sample
  STRM of CYTS of type LLLLLfffBB with 4 samples -- 10 elements per sample
  STRM of CSEN of type LffffffLLLL with 4 samples -- 11 elements per sample

MP4 Payload time 0.000 to 1.001 seconds
GPRI 145.912s, 33.125deg, -117.331deg, 15.445m, 0.250m, 0.420m, 0.590m/s, 334.350deg, 3.000, 20.000, 
GPRI 146.312s, 33.125deg, -117.331deg, 15.367m, 0.250m, 0.420m, 1.240m/s, 325.640deg, 3.000, 20.000, 
GPRI 146.413s, 33.125deg, -117.331deg, 15.363m, 0.250m, 0.420m, 1.500m/s, 328.710deg, 3.000, 20.000, 
GPRI 146.710s, 33.125deg, -117.331deg, 15.331m, 0.250m, 0.420m, 2.510m/s, 330.990deg, 3.000, 20.000, 

ACCL sampling rate = 192.549209 Hz (from -0.135230 to 12.313536)
GYRO sampling rate = 385.114885 Hz (from -0.136323 to 12.314507)
ISOG sampling rate = 29.047875 Hz (from -0.113870 to 12.313889)
SHUT sampling rate = 29.047875 Hz (from -0.113870 to 12.313889)
FWVS sampling rate = 7.728535 Hz (from -0.159250 to 12.262250)
KBAT sampling rate = 0.966067 Hz (from -0.159250 to 12.262250)
GPRI sampling rate = 3.803888 Hz (from -0.050556 to 12.305222)
ATTD sampling rate = 3.803888 Hz (from -0.050556 to 12.305222)
GLPI sampling rate = 3.803888 Hz (from -0.050556 to 12.305222)
VFRH sampling rate = 3.803888 Hz (from -0.050556 to 12.305222)
SYST sampling rate = 0.966067 Hz (from -0.159250 to 12.262250)
BPOS sampling rate = 0.905688 Hz (from -0.806867 to 12.442733)
ATTR sampling rate = 3.803888 Hz (from -0.050556 to 12.305222)
SIMU sampling rate = 3.803888 Hz (from -0.050556 to 12.305222)
ESCS sampling rate = 0.966067 Hz (from -0.159250 to 12.262250)
SCPR sampling rate = 0.490026 Hz (from 0.413947 to 12.658188)
LNED sampling rate = 3.803888 Hz (from -0.050556 to 12.305222)
CYTS sampling rate = 3.864268 Hz (from -0.159250 to 12.262250)
CSEN sampling rate = 3.781932 Hz (from -0.111868 to 12.315642)
=================================================================
==63528==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x61f000000c6c at pc 0x5601cc3c88fa bp 0x7fff24d2cb30 sp 0x7fff24d2cb20
READ of size 4 at 0x61f000000c6c thread T0
    #0 0x5601cc3c88f9 in GPMF_Next /home/tim/gpmf-parser-asan/GPMF_parser.c:300
    #1 0x5601cc3c9163 in GPMF_FindNext /home/tim/gpmf-parser-asan/GPMF_parser.c:344
    #2 0x5601cc3ee290 in main /home/tim/gpmf-parser-asan/demo/GPMF_demo.c:245
    #3 0x7f3d5aad3b6a in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x26b6a)
    #4 0x5601cc3c6299 in _start (/home/tim/gpmf-parser-asan/build/gpmf-parser+0x2299)

0x61f000000c6d is located 0 bytes to the right of 3053-byte region [0x61f000000080,0x61f000000c6d)
allocated by thread T0 here:
    #0 0x7f3d5ada487e in __interceptor_realloc (/lib/x86_64-linux-gnu/libasan.so.5+0x10c87e)
    #1 0x5601cc3ee6f9 in GetPayload /home/tim/gpmf-parser-asan/demo/GPMF_mp4reader.c:64
    #2 0x5601cc3ed611 in main /home/tim/gpmf-parser-asan/demo/GPMF_demo.c:102
    #3 0x7f3d5aad3b6a in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x26b6a)

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/tim/gpmf-parser-asan/GPMF_parser.c:300 in GPMF_Next
Shadow bytes around the buggy address:
  0x0c3e7fff8130: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c3e7fff8140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c3e7fff8150: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c3e7fff8160: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c3e7fff8170: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c3e7fff8180: 00 00 00 00 00 00 00 00 00 00 00 00 00[05]fa fa
  0x0c3e7fff8190: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c3e7fff81a0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c3e7fff81b0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c3e7fff81c0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c3e7fff81d0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==63528==ABORTING
```

dnewman-gpsw, 2019-10-17:
I can't reproduce this to failure. Please confirm with the develop branch.

cuanduo, 2019-10-18:
I have tried develop version,the result is same.have you compiled with asan.or unzip the zip file
```
root@ubuntu:/home/tim/gpmf-parser-develop/build# git log
commit 68c9572b51855b0b45afba48787a1b301fa0350e (HEAD -> develop, origin/develop)
Author: Teeto Cheema <tcheema@gopro.com>
Date:   Wed Oct 16 11:17:12 2019 -0700

    Fix for stsz table condition when the sampleSize is non-zero.

commit 644907c739868ccdfb933b36749e066b64272d97
Author: David <dnewman@gopro.com>
Date:   Mon Oct 7 15:56:15 2019 -0700

    Added an API for reading the Edit List offset

commit c92fdd3ec6f1daa44ff693f2b008a44be72f8142
Merge: 1d77465 33f30e5
Author: David Newman <dnewman@gopro.com>
Date:   Mon Oct 7 10:53:55 2019 -0700

    Merge pull request #73 from gopro/Fix-GetPayloadRationalTime
    
    Fix GetPayloadRationalTime to match numbers given by GetPayloadTime

commit 1d7746590d291195fda5919ad837ccfa19510b8d
Author: David <dnewman@gopro.com>
Date:   Fri Oct 4 11:52:28 2019 -0700

    Oopps! Removed left off #endif

commit 33f30e578aec9e3706afa4f80abf4fc2524d2a6a
Author: Teeto Cheema <tcheema@gopro.com>
Date:   Fri Oct 4 11:44:38 2019 -0700

    Fix GetPayloadRationalTime to match numbers given by GetPayloadTime

commit a895c964b1913eef2de48650a526c2dd3561da5a
Author: David <dnewman@gopro.com>
Date:   Fri Oct 4 11:23:13 2019 -0700

    very minor fix debugging using PRINT_MP4_STRUCTURE

commit 2fee9f54b00613f6089aa9fc3f2e73541185e9a9
Author: David <dnewman@gopro.com>
Date:   Fri Sep 27 10:48:10 2019 -0700

    Fixed for handling stts with all entries at duration 1

commit eec612b3ca184d867ff9671ac0119e30293ffb24
Author: David <dnewman@gopro.com>
Date:   Thu Sep 26 08:51:14 2019 -0700

    removed redundant and clashing define

commit c058410217c8e8366a61f14eb92cde3cff6c9646
Author: David <dnewman@gopro.com>
Date:   Wed Sep 25 13:49:15 2019 -0700

    fixed for edit lists and demo for precision video to metadata sync

commit 91fceedd1c5e3e324c9c131146a469f03d7169ed
Merge: 4e17078 c102223
Author: David Newman <dnewman@gopro.com>
Date:   Wed Sep 4 07:33:00 2019 -0700

    Merge pull request #72 from G-P-S/ylaala/fix-negative-offset
    
    mp4reader: switching in_numerator & out_numerator to int32_t
root@ubuntu:/home/tim/gpmf-parser-develop/build# ./gpmf-parser karma.mp4-out_of_bound-idx\:0x24-0x0 
video framerate is 29.97 with 362 frames
[... 264 lines cut ...]
  0x0c3e7fff81c0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c3e7fff81d0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==52138==ABORTING

```

dnewman-gpsw, 2019-10-18:
I have not changed my compiler, as I figure the failure would be general, but I did use all the provided zip files, and I get no failure.  I can see that the files are corrupted, but the parser correctly handles those corruptions.  Maybe you could point out which line of code begins the heap overflow for each POC and what you would change?

cuanduo, 2019-10-18:
you should compile with cflags +=-fsanitize=address,or some heap-overflow may not result in segment fault.

and code line is showned in asan info
```
 GPMF_Next /home/tim/gpmf-parser-develop/GPMF_parser.c:302

294             if (ms->pos < ms->buffer_size_longs)
 295             {
 296                 while (ms->pos < ms->buffer_size_longs && ms->nest_size[ms->nest_level] > 0 && ms->buffer[ms->pos] == GPMF_KEY_END)
 297                 {
 298                     ms->pos++;
 299                     ms->nest_size[ms->nest_level]--;
 300                 }
 301 
 302                 key = ms->buffer[ms->pos];
 303                 if (!GPMF_VALID_FOURCC(key))
 304                     return GPMF_ERROR_BAD_STRUCTURE;
 305 
 306                 if (key == GPMF_KEY_DEVICE_ID)
 307                     ms->device_id = BYTESWAP32(ms->buffer[ms->pos + 2]);
 308                 if (key == GPMF_KEY_DEVICE_NAME)

```

dnewman-gpsw, 2019-11-18:
size field corruption fixed.

closed 2019-11-18 by dnewman-gpsw