gopro/gpmf-parser #80: STMP precision

JuanIrache, 2019-10-27:
Hi, loving the new streams and additions!

How should the STMP values be interpreted? Are they the time stamp of the first sample in a STRM chunk? The last one? The average for the chunk?

I read in another issue that STMP is the time the data was sent, so maybe that means when all the samples in the chunk have been gathered.

Thanks!

dnewman-gpsw, 2019-10-27:
STMPs are the computed timestamp for the first sample in each payload.  As each stream can be from a different clock source, the precision can vary.  All video metadata sources the precision is excellent: CORI, IORI, GRAV, SHUT, ISOE, HUES etc.  Data like ACCL & GYRO are still needing work to improve the phase precision (I think is about 10ms delayed in the modes I've tested.)  However ACCL & GYRO are less important compared with the sensor fusion data like CORI (Camera Orientation), IORI (Image Orientation) and GRAV (Gravity vector). All the MAX and HERO8 horizon leveling and post stabilization effects are now based solely on these new streams.  As these new streams are one sample per frame, you don't need to use STMP.  The STMP was not intended for post synchronization, as it has already been used for in-camera synchronization.  It is not provided in the TLV and Timewarp modes, and as the in camera sync improves, it may be removed in future.  Of course, while it is useful it will remain.

JuanIrache, 2019-10-27:
Thank you for the very comprehensive explanation

closed 2019-10-27 by JuanIrache

jpilet, 2023-02-20:
This explanation should go in docs/README.md. It is super useful.