gopro/gpmf-parser #81: Hero8 samples not representative?

JuanIrache, 2019-12-01:
Hi,

I noticed the data in the Hero 8 sample is different from what real users have at the moment. For example the -I presume- newest firmware records whether the GPS altitude is mean sea level or otherwise (GPSA), which is very useful information. However, while the sample (https://github.com/gopro/gpmf-parser/tree/master/samples) can be loaded in GoPro's Quik software (which does not support the new cameras yet) by just modifying the DVNM to an old one (like "Hero7 Black"), when real users try to do that with real videos, the gauges do not show. I know Quik is not strictly related to this repository, but maybe having up-to-date samples would be useful in order to adapt our software to the latest files.

Or maybe I'm looking in the wrong place and the new data has nothing to do with Quik not displaying stickers.

Thanks

dnewman-gpsw, 2019-12-01:
The HERO8 sample posted is from release firmware (v1.20.) The new v1.50 was release a few days after the new samples where up loaded, so this sample will still be representative of most HERO8 clips to date.  GPSA was added in v1.50, previously all the GPS Enabled cameras used WGS 84, for altitude above the ellipsoid.  Yes, altitude using Mean Sea Level is better.  I'm sure Quik's support for all new camera features are here or pending very soon.  FYI: All of these clips are the LRV renamed to MP4, to keep the file small, however the metadata is identical.

JuanIrache, 2019-12-01:
Thank you. That makes sense.

Would you say GPSA could be the reason Quik is not reading the data? Or is there some other major change between v1.20 and v1.50, metadata-wise?

dnewman-gpsw, 2019-12-01:
I don't directly follow Quik issues, particularly as most the newest editing services have moved in the GoPro App.  Do the gauges to work in the GoPro App?  I doubt GPSA would be the issue.

JuanIrache, 2019-12-01:
Not sure if they work on the GoPro app, as users send me their videos asking for assistance, and the Android app does not allow importing external files. Is a new GoPro App coming for desktop?

Another small change worth mentioning is that DVNM has changed from size: 1 repeat: 11 to size: 11 repeat: 1. But changing that back does not enable gauges either.

I will close this as it is not very relevant to the GPMF parser, I guess. Thanks agian.

closed 2019-12-01 by JuanIrache