gopro/gpmf-parser #82: GoPro MAX magnetometer usage

y3nd, 2019-12-01:
Hi,

I'm currently trying to read magnetometer data and compute heading from the x and y values but somehow the "common heading formula" (https://www.w3.org/TR/magnetometer/#compass) doesn't give the right results (even by using a -Y,X,Z or Y,-X,Z format).

A raw value read from GPMF like this one `[-112, 27, 141, 6738]` should give a heading of about 26 degrees (if north is 0 degree and east is 90 degrees).

Is it possible to compute the heading without GPS data interpolation to get an absolute compass when the camera is not moving/underwater ?

dnewman-gpsw, 2019-12-01:
The bug has been filed with the camera firmware, the magnetometer is not correct.

y3nd, 2020-02-18:
Do you have any ETA on a fix ? Is it even fixeable on the camera firmware or hardware-related ?

dnewman-gpsw, 2020-02-25:
It is an open firmware issue that is being investigated.  On Fusion we computed the 3-axis µT data using the calibration data stored with the BMI150 magnetometer.   On MAX are have stored, the raw data, three axis x,y,z and rhall (4 raw values, rather than 3 axis computed values.)  From this data µT values can be computed, but accurately can only be achieved with the calibration/trim data that is missing. No ETA of any change for this, can you use the data as is?

dnewman-gpsw, 2021-03-01:
This is now corrected in v2.0 firmware for MAX.

closed 2021-03-01 by dnewman-gpsw

arasharchor, 2022-04-13:
@dnewman-gpsw 

has GoPro 7 black Magnetometer? How can I access it if it has this sensor?