gopro/gpmf-parser #83: GoPro MAX GPS

trackdaygenius, 2020-01-21:
Hi David it's been a while, I have a question, we need clarification regarding the GPS data available from the GoPro MAX 360.

So, can we expect the same GPS data output from the MAX360 as we have come to expect from the current HERO series cameras? 

We're running into some unexpected GPS data anomalies, whether we've received a defect camera or there are some inbuilt limitations?  

Our motorcycle tester was on track at the Sepang circuit in Malaysia, but the GPS data read:
gpsFix = 0 means no satellite locked - this also means no gps coordinates. It simply used the last known gps location which is our office in London  - Edgware Rd.

Thanks

kofi

dnewman-gpsw, 2020-01-21:
That is not typical. It should behave like other HERO cameras.

shichonghuotian, 2020-01-22:
Hi David,
I used this [demo](https://github.com/gopro/gpmf-parser/blob/master/demo/GPMF_demo.c) to test .360 videos，I found that GPS data is the same。

Video link: [video](https://www.dropbox.com/s/wfw38za2gh28tnx/GS010009.360?dl=0)

The log：
MP4 Payload time 136.136 to 137.137 seconds
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s, 
GPS5 51.520deg, -0.167deg, 109.186m, 0.000m/s, 0.000m/s,

dnewman-gpsw, 2020-01-22:
Also GPSU says the GPS date-time is 2015-10-18-00:00:00, which suggests the date and time was never read from any GPS lock. Why this is I can't determine from the data.

closed 2020-01-27 by dnewman-gpsw