gopro/gpmf-parser #84: Best practices for data sync

hit2sjtu, 2020-03-02:
I was aware that the timing between gyroscope/accelerometer and images may not sync well. I was wondering what is the best practice to sync these? Is STMP needed? I am having both hero 7 white and max.

dnewman-gpsw, 2020-03-02:
On MAX the sync is very good, and it internally uses STMP time stamps to keep the IMU data in phase with the video.  You would only use STMP if you wanted sub-frame level precision.  For HERO7 White, I have never tested one of those, as I only personally review the Black and other premium cameras like MAX and Fusion.  I expect that entry level HERO7 is still pretty accurate.

hit2sjtu, 2020-03-03:
Thanks for the clarification. I only tested hero 7 white and the initial results are not so good. I will do some comparison with MAX to see if it is better.

closed 2020-03-04 by hit2sjtu