gopro/gpmf-parser #85: Unable to process larger file sizes

BeachWeasel, 2020-03-03:
OpenMP4Source() in GPMF_mp4reader.c returning '0' for larger mp4 files.

dnewman-gpsw, 2020-03-03:
What size?

BeachWeasel, 2020-03-04:
Thanks for the quick response.
~3GB. I can share it tomorrow if you would like. Just let me know.

dnewman-gpsw, 2020-03-04:
Weird, that would surprise me.  GoPro mp4 chapters are just under 4GB, and this tools is used on those all the time.  There might be sometime else at issue, not the file size itself.  Share a link file, I will help determine the issue.

BeachWeasel, 2020-03-04:
AUDIO WARNING! 
These were recorded with another system that embeds metadata into the left audio channel of the video. You may need to turn down or mute your audio if you play these video files.

2.52GB, Not sure which Hero Model this was recorded on, should have GPS data, returns 0 for mp4stat.st_size:
https://mega.nz/#!AUMgWAbK!Ad5n_0tyIB4UaEkKa3VL1RmLlJXKpVo6_B5N9ee7Fxc

Also, I have a couple of hero4 videos that have different results from mp4stat.st_size in GPMF_mp4reader.c:

3.73GB, Hero4, returns 0 for mp4stat.st_size:
https://mega.nz/#!cMMQQYjD!QXcVDYiR0XqUAlyJ5G-D6U787z-y0uQga2edjrK9MGM

1.7GB, Hero4, recorded on same camera above, does not return 0 for mp4stat.st_size:
https://mega.nz/#!IRNSjIpJ!8AUTqWwx11w0B1hRbMAGsQ4XU78Wv2UuMTCqfmJpbf8

Thanks. I really appreciate your work. I don't usually work with the c language.

dnewman-gpsw, 2020-03-04:
All those files are behaving correctly.  The 2.5GB clip is from HERO8, and the parser reads the GPMF correctly.  The two HERO4 clips correctly return 0 as HERO4 was prior to GPMF being added to GoPro cameras.

dnewman-gpsw, 2020-03-09:
A VS2017 solution file and vcxproj
[build.zip](https://github.com/gopro/gpmf-parser/files/4308002/build.zip)

closed 2020-03-13 by dnewman-gpsw

BeachWeasel, 2020-03-13:
That worked. Thank you very much!