gopro/gpmf-parser #86: command line tools for go max format

hit2sjtu, 2020-03-05:
I am not sure if this is the right place to ask this question. But I am looking for some help here.
For gopro fusion, I am aware there is command line tools to extract mp4 video
https://www.studentfilmmakers.com/the-secret-to-extracting-5-7k-video-from-the-gopro-fusion/

Is there any way to export .mp4 file from .360 file for GoPro max from command line? Windows MacOS or linux ?

dnewman-gpsw, 2020-03-05:
The .360 file is an MP4 file, just renamed.  Within the MP4 is a two strip EAC projection. Tech info here https://gopro.com/en/us/news/max-tech-specs-stitching-resolution  You are likely really asking can you have shell tool to convert from EAC to ERP?  Yes, it is possible, but it doesn't exist yet, as I don't think the GoPro Exporter or GoPro Player can be run from the command line.  https://gopro.com/help/articles/block/GoPro-Player No, this is not right place to ask a non-GPMF-parser question.  Try https://community.gopro.com/

closed 2020-03-05 by dnewman-gpsw

hit2sjtu, 2020-03-05:
Thanks. That is all the info I needed.