gopro/gpmf-parser #88: Bluetooth devices Hero8

JuanIrache, 2020-03-30:
Hello,

Are external bluetooth devices not supported any more on the Hero8? The connection interface seems to have disappeared from the mobile app (It's still there if using an older model).

Thank you

dnewman-gpsw, 2020-03-30:
I don't know.  This is the first I've heard of this.  I will ask.

dnewman-gpsw, 2020-03-30:
It is still enabled on camera, I don't know if or when it will be re-enabled in the App.  I hope there is a good reason for this, but there is nothing I can disclose.

closed 2020-03-30 by dnewman-gpsw

Gangti, 2020-09-27:
> It is still enabled on camera, I don't know if or when it will be re-enabled in the App. I hope there is a good reason for this, but there is nothing I can disclose.

Hi, I googled GoPro bluetooth related topics, and tried shutter control, and it did worked with my GoPro Hero8 Black.
The problem is that I don't know the data format when I try sending sensor data to GoPro with following UUID.
b5f90078-aa8d-11e3-9046-0002a5d5c51b SensorData
b5f90079-aa8d-11e3-9046-0002a5d5c51b SensorDataResponse

Could you provide some details on the above UUID usage?

dnewman-gpsw, 2020-09-27:
GoPro is working on new documentation for this. Nothing I can share today.