gopro/gpmf-parser #89: GPS time discrepancy between GoPro's

chrisdenley, 2020-04-12:
I've been trying to sync videos from multiple GoPro's based on GPS time, but it is always off by up to a second. I wasn't able to use your library as I'm not comfortable with C and I had issues even compiling the demo code (not enough disk space for MS VS, MinGW won't compile it, Linux failed because the code is 32-bit, etc). I wrote my own Perl code to parse the GPMF data. The GPMF track for either camera has no ctts or edts, so determining the playback time of the GPS payload is pretty straight-forward with the stts.

However, as demonstrated in this video, the GPS timecodes are not consistent between the Hero 7 Black and the Max. Do you have any idea what the issue is? Is there an offset or delay that is not included in the track data? Can you provide code that will map GPS timestamps to playback time that I may be able to compile so I can validate it matches my output? I suspect it is an issue with the GoPro's rather than my code. It obviously isn't an issue with your gpmf-parser code (I'm not using it), but I would appreciate any assistance.

https://youtu.be/IdgaiVW0mbw

dnewman-gpsw, 2020-04-12:
Weird on your compiling issues, the code is tiny and not 32-bit specific.  The GPS clock issue, I guessing your MAX is brand new, and hasn't been receiving a good GPS look for more than 5 minutes of its entire running life. GPS required leap seconds to be transmitting from the satellite, normally once every few minutes, once the camera see the leap second signal it will be correct until the next leap second change (every few years.) https://racelogic.support/01VBOX_Automotive/01General_Information/Knowledge_Base/What_are_GPS_Leap_Seconds

chrisdenley, 2020-04-12:
It is brand new as it was replaced under warranty a few weeks ago, but it was recording outside last weekend for about an hour, and surely must've had a GPS lock for 5 minutes. In fact, I just verified that it had a GPS lock the whole time (beginning at 1.001 seconds in) from 200404211907.630 to 200404221348.654, so nearly an hour straight. Then it recorded again for about 15 minutes. Also, I noticed this discrepancy before it was replaced under warranty as well.

dnewman-gpsw, 2020-04-12:
Your own proves this camera has never seen a GPS Lock, notice that the date and time is 151018000000, 2015 Oct 18th, midnight.  That is a newly born GPS clip.  Once a lock is established to latest known date, time and location is save to non-violate memory, so that the GPS looks faster on the next power on.  I do expect in the video you showed me that the MAX hadn't yet received it leaps seconds signal.  

All of that aside, there is no design (yet) that will allow GPSU date/time to frame-sync across camera models.  GPSU is written only once per payload (1.001s typically) and it is not timestamped, so it is only good of about 1 second between cameras of the same model.  The GPSU value is not guaranteed to match any particularly video frame.  Camera with different model types will have differing pipeline delays, so that could throw you off another second.  Two H7s or two MAXs, shoot in the same mode, you would be closer. 

It has been considered to timestamp GPSU, so it could be used for sync, but that is not enabled in any camera model to date.

chrisdenley, 2020-04-12:
I told you the GoPro Max was recording from 200404211907.630 to 200404221348.654 with a GPS lock indicated in GPMF each payload. It was the same GoPro Max as I received the replacement on 4/2 two days prior to the creation of that video and can provide the UPS tracking number or GoPro order number. I have only ever had one Max in my possession at any given time. I have further proof in over 25 GB of unstitched 360 video with GPMF data. So if my video supposedly proves it has never seen a GPS lock, then how come it never saw a GPS lock in that time? I thought it should only take a few minutes. Do I have yet another defective unit?

It is disappointing to learn that I cannot count on the GPS data to be accurate within a second. It seems like a missed opportunity as it would've been quite useful, but I suppose it is not a very typical use case. Better to fine-tune the sync manually to compensate for GoPro's inaccuracy than starting from scratch I guess.

dnewman-gpsw, 2020-04-12:
The clip you shared shows an uninitialized GPS, indicated with the 2015 date it reported (in your video.) Maybe you shared the wrong clip? That is all I have to go on. In this clip gained at least one satellite lock during the recording, you see the date switch from 2015 to 2020.  Within the next 5 minutes it would likely gain a leap second update. There is nothing wrong with the camera, video again shows it is working as expected.  I do not work in support, I can only answer questions about GPMF.

chrisdenley, 2020-04-13:
I don't know how you can say there is nothing wrong with the camera while also saying that should not be possible. No, I didn't share the wrong clip. The GPS should've been initialized already last weekend. You can even see the edited video at the link below which contains footage of and from the GoPro Max. It was posted to YouTube before the test footage was even recorded.

https://www.youtube.com/watch?v=EHIjvcgsWyI

I guess I can keep testing to see if it ever does function the way you say it should, and if it keeps giving me a 2015 date at the start of the recording, it must be yet another defect.

closed 2020-04-13 by chrisdenley