gopro/gpmf-parser #91: GoPro 360 video projection type

kuku131, 2020-04-30:
Hi David,

Hope you are well!

I have a quick question that would appreciate for your help on.

We are working on GoPro 360 video integration and are trying to play GoPro .360 videos within our app. However we noticed that there are 3 projection types:
1 - GoPro lens
2 - Equirectangular
3 - Equal Area cube map

Can you please advise how to obtain the projection type from video metadata, so we can enable the right mode to play the video?

Many thanks!

Xiaoyue Huo

dnewman-gpsw, 2020-05-01:
The GPMF blob with the header ('udta' atom) of .360 (MP4) file, has the tag PRJT (Projection).  Samples  values: GPRO – stock GoPro Lens,  ERP0 – ERP Projection, EAC0 – Stock EAC0 (currently not used in latest firmware) and EACO – EAC Overlap (used now.)

kuku131, 2020-05-01:
Hi David,

Thanks very much for your reply! Much appreciated.

We will look into it and come back if we have more questions.

I wish you have a lovely weekend!

Regards,
Xiaoyue Huo

closed 2020-05-01 by kuku131