gopro/gpmf-parser #92: Play GoPro .360(mp4) file directly using our app

kuku131, 2020-05-07:
Hi David,

I hope you are well!

I have a quick question that would appreciate for your help on:

We analyzed the GoPro raw .360(mp4) file and tt looks like there are 2 video tracks thus it can't be played directly using a standard 360 video player.
Currently, we have to use GoPro app to process its projection type to Equirectangular before it can be played properly within our app.
May I know if there is any way to avoid having to use GoPro app to change its projection type and play the raw .360(mp4) file directly within our app? 

Many thanks!

Xiaoyue Huo

dnewman-gpsw, 2020-05-07:
Some info here. https://gopro.com/en/ch/news/max-tech-specs-stitching-resolution  The video was intentionally split into two for improved mobile compatibility and performance, so it is not stock EAC.  Other changes also make is harder for third parties to reproject with EAC math alone, although not impossible. There is a band of duplicated pixels that help us do more advanced filtering that the camera can't do, to improve the exposure blend between the lens. Unfortunately the details are not public, nor do I have them, but I'm sure you can work them out.

trackdaygenius, 2020-05-08:
Hey David, 

Thanks for the assist, this particular area of knowledge is beyond our skills set.  We appreciate your help over the years, was fun being part of the now defunct Dev program, here's the results of our efforts:

- https://www.youtube.com/watch?v=iNodLSSjQ48&t=125s

Insta360, OSMO and Garmin are nipping at your heals, keep up the good work, hopefully GoPro will resurrect the Dev program one day, community is important.

Thanks

kofi

dnewman-gpsw, 2020-05-08:
Wow! Very very good. Did you do the .360 decode and blend? 

As for Dev Program, I hope it replaced with something better (can't say or promise.)

trackdaygenius, 2020-05-08:
the .360 decode is beyond our skills set, customers are having to engage in the horrible UX of exporting from the GoPro app (assuming the GPS data isn't ignored), then import into the TrackDayGenius app, then re-export with our circuit data and performance overlays, but the results are superb when customers don't abandon the process :(

dnewman-gpsw, 2020-05-08:
Okay.  While full res .360 decoding is not that hard, the average playback engine is not expecting dual streams.  This is required for any greater than 4K 360 decoding, even from our competitors. Hopefully that will get easier.

closed 2021-03-01 by dnewman-gpsw