gopro/gpmf-parser #95: HERO8 sample's GPS data seems incorrect

ryan-williams, 2020-06-20:
[I changed the demo tool to output `GPS5` and `GPRI` streams (instead of the default `ACCL` stream)](https://github.com/ryan-williams/gpmf-parser/commit/57079ce806271d4f6264545b0a3e44a7c063fa99), and ran it on the HERO8 sample file:

```bash
 gpmfdemo ../samples/hero8.mp4
```
<details>
<summary>Click to show output</summary>

```
video framerate is 29.97 with 379 frames
  STRM of ACCL of type s with 202 samples -- 3 elements per sample
  STRM of GYRO of type s with 202 samples -- 3 elements per sample
  STRM of SHUT of type f with 30 samples
  STRM of WBAL of type S with 9 samples
  STRM of WRGB of type f with 9 samples -- 3 elements per sample
  STRM of ISOE of type S with 30 samples
  STRM of YAVG of type B with 9 samples
  STRM of UNIF of type f with 9 samples
  STRM of SCEN of type Ff with 9 samples -- 2 elements per sample
  STRM of HUES of type BB with 9 samples -- 2 elements per sample
  STRM of FACE of type Lffffff with 30 samples -- 7 elements per sample
  STRM of GPS5 of type l with 19 samples -- 5 elements per sample
  STRM of CORI of type s with 29 samples -- 4 elements per sample
  STRM of IORI of type s with 29 samples -- 4 elements per sample
  STRM of GRAV of type s with 29 samples -- 3 elements per sample
  STRM of WNDM of type BB with 10 samples -- 2 elements per sample
  STRM of MWET of type BBB with 10 samples -- 3 elements per sample
  STRM of AALP of type bb with 10 samples -- 2 elements per sample

MP4 Payload time 0.000 to 1.001 seconds
GPS5 42.027deg, -129.294deg, 9540.240m, 0.000m/s, 0.000m/s,
GPS5 42.026deg, -129.294deg, 9540.182m, 0.000m/s, 0.000m/s,
GPS5 42.026deg, -129.294deg, 9540.123m, 0.000m/s, 0.000m/s,
GPS5 42.578deg, -129.517deg, 9745.102m, 0.000m/s, 0.000m/s,
GPS5 42.578deg, -129.517deg, 9745.046m, 0.000m/s, 0.000m/s,
GPS5 42.026deg, -129.293deg, 9539.947m, 0.000m/s, 0.000m/s,
GPS5 42.026deg, -129.292deg, 9539.889m, 0.000m/s, 0.000m/s,
GPS5 42.578deg, -129.516deg, 9744.876m, 0.000m/s, 0.000m/s,
GPS5 42.578deg, -129.515deg, 9744.820m, 0.000m/s, 0.000m/s,
GPS5 42.265deg, -129.388deg, 9628.708m, 0.000m/s, 0.000m/s,
GPS5 42.265deg, -129.388deg, 9628.650m, 0.000m/s, 0.000m/s,
GPS5 42.265deg, -129.387deg, 9628.593m, 0.000m/s, 0.000m/s,
GPS5 41.705deg, -129.163deg, 9421.185m, 0.000m/s, 0.000m/s,
GPS5 41.705deg, -129.163deg, 9421.125m, 0.000m/s, 0.000m/s,
GPS5 38.433deg, -127.929deg, 8220.372m, 0.000m/s, 0.000m/s,
GPS5 38.433deg, -127.929deg, 8220.301m, 0.000m/s, 0.000m/s,
GPS5 38.433deg, -127.928deg, 8220.230m, 0.000m/s, 0.000m/s,
GPS5 38.041deg, -127.788deg, 8078.376m, 0.000m/s, 0.000m/s,
GPS5 37.665deg, -127.655deg, 7942.678m, 0.000m/s, 0.000m/s,

ACCL sampling rate = 202.465821Hz (time -0.001103 to 12.677580)",
GYRO sampling rate = 202.465821Hz (time -0.001103 to 12.677580)",
SHUT sampling rate = 29.970035Hz (time 0.000000 to 12.645965)",
WBAL sampling rate = 9.990010Hz (time 0.000000 to 12.612600)",
WRGB sampling rate = 9.990010Hz (time 0.000000 to 12.612600)",
ISOE sampling rate = 29.970035Hz (time 0.000000 to 12.645965)",
YAVG sampling rate = 9.570262Hz (time 0.000000 to 12.643332)",
UNIF sampling rate = 9.990010Hz (time 0.000000 to 12.612600)",
SCEN sampling rate = 9.570262Hz (time 0.000000 to 12.643332)",
HUES sampling rate = 9.570262Hz (time 0.000000 to 12.643332)",
FACE sampling rate = 29.970035Hz (time 0.000000 to 12.645965)",
GPS5 sampling rate = 18.173129Hz (time -0.063608 to 12.647467)",
CORI sampling rate = 29.970031Hz (time 0.000000 to 12.612600)",
IORI sampling rate = 29.970031Hz (time 0.000000 to 12.612600)",
GRAV sampling rate = 29.970031Hz (time 0.000000 to 12.612600)",
WNDM sampling rate = 10.000000Hz (time -0.081672 to 12.618328)",
MWET sampling rate = 10.000000Hz (time -0.081672 to 12.618328)",
AALP sampling rate = 10.000000Hz (time -0.081672 to 12.618328)",
```

</details>

Those `GPS5` values seem bogus to me; they move too much from start to finish (for a video that apparently shows a person walking for about 10s), and the coordinates are all [somewhere in the Pacific ocean ≈200mi off the coast of California](https://goo.gl/maps/xkgooad9ASs8U6817).

I also ran the demo on a video I took while stationary in NYC, and it at least showed one GPS value (`46.794deg, -42.711deg`) for all frames, but [those coordinates are in the middle of the Atlantic, ≈1000mi from where the video was filmed](https://goo.gl/maps/vnmRyVbjfTjmAmgUA).

I've already [observed gauges not to be available in Quik](https://community.gopro.com/t5/en/How-to-Use-Gauges-in-Quik-for-Desktop/ta-p/401626):

[image]

I also see that [the GoPro "Telemetry Extractor" sells (for $50) a piece of software that claims to fix HERO8 GPS and make it work in Quik](https://goprotelemetryextractor.com/gopro-hero-8-no-gps-data).

Are there other known issues with the GPS telemetry on HERO8's?

Is there some other way I can get at this data?

ryan-williams, 2020-06-20:
Running the same demo on the HERO7 sample gives only lines like:
```
GPS5 0.000deg, 0.000deg, 0.000m, 0.000m/s, 0.000m/s,
```

The HERO5 and HERO6 samples seem to give reasonable data:
```
GPS5 33.126deg, -117.327deg, -20.184m, 0.167m/s, 0.190m/s,
```
```
GPS5 33.127deg, -117.327deg, -17.228m, 1.221m/s, 1.230m/s,
```
[both seem point to what looks like an office park outside San Diego](https://goo.gl/maps/cdHFYte4mH96CESG9).

dnewman-gpsw, 2020-08-20:
I apologize for the delay.  The HERO8 example was not with a GPS lock, this is why the coordinates are invalid.  The GPSP and GPSF fields both indicate that lock is bad.

closed 2020-08-20 by dnewman-gpsw