gopro/gpmf-parser #100: Coordinate systems of CORI and GRAV

Kjue, 2020-07-09:
This is a great interface in and of itself already, I'm trying to figure out how to use values I'm getting from my GoPro MAX at the moment.

I have access to the CORI, IORI and GRAV values of the stream, and I am trying to orient the image to the world coordinates so it would stay stabilized. I have a timelapse material that seems to be at an FPS of 6.

My target 3D system has right handed coordinates where in XYZ-coordinates the Y-axis is the up direction. My end goal is to have the same effect as the MAX exporter has for **world lock** and **horizon level**.

1) What handed system are the vec3 stored as?
2) How about the quaternions?
3) What are the orders of the data for CORI, IORI and GRAV streams?
    - I noticed that GYRO is having data ordered as Y,-X, Z.

Any help is appreciated, thank you!
Mikael

babindemers, 2020-07-09:
I had the same problem a few days ago. When you look at the picture with the coordinate system in the README, the Z axis is up, and the Y axis is pointing backwards. To access the data with this coordinate system, the ACCL and GYRO need to be read with Z, X and Y order. For CORI, I extracted the quaternion with [q0, qx, qz, qy]. 

It doesn't correspond to the order specified in the documentation, but I find it easier to start from the coordinate system on the picture and then do the transform.

I haven't worked with the GRAV stream yet, so I can't help you with that.

Hope this helped!

Kjue, 2020-07-10:
Thanks @babindemers, yeah that helped a bit. I also figured out a way of helping myself with the material. I will report back if I make some progress. Still stumped at the moment a bit, and will try to spend some time in the coming week. I figured that I can add visualisations on the output to help me illustrate if I'm in the right ballpark with the transforms.

Kjue, 2020-07-15:
Right, coming back to this now that it makes some sense to me. The data in the quaternions is ordered as **W,X,Z,Y** from the data itself as ordered values. The rotations are of a right handed system as described on the documentation [here](https://github.com/gopro/gpmf-parser#where-to-find-gpmf-data) and shown on the picture. I tested it.

[image]

And looking at the gravity vector now from the same point of view, I do not see them sharing the same axis as gyroscope does. If I were to label these from the same axis labels as on the image, they would be **X,Z,Y**.

I am not however fully satisfied that the quaternion is stabilized properly and I think that there is some magic happening for the actual stabilization still. However it is remarkably good starting point!

I feel like I've now achieved the main part of the features sufficiently. I would appreciate some insights how to get the video really stable when just viewing the content so they appear more smooth.

Any help is still appreciated or even insights. Thank you!

closed 2020-07-16 by Kjue

takerfume, 2022-06-22:
@Kjue 
Hi.
Could you please tell me how to realize world lock and horizon leveling with COIR and gravity vector?

I am glad if you share some codes or numerical formula.

Kjue, 2022-06-26:
@takerfume Hi, I wasn't ever really able to achieve a proper **horizon lock** with those details, only the **stabilization part** so that the image was stable through the movements and would not bounce around crazily. The final piece is the horizon lock so that everything really levels out but that part I was not able to achieve. I'll be curious to learn if you come up with something specific to that.

takerfume, 2022-07-11:
> @takerfume Hi, I wasn't ever really able to achieve a proper **horizon lock** with those details, only the **stabilization part** so that the image was stable through the movements and would not [...]

thank you! I will try it!

Kjue, 2023-06-21:
@krishbhk Hi! My use case was also specifically MAX only. And my findings were especially for that camera.

I simply tested it out by having a mechanism through gpmf-reader to see and evaluate the values, then used the [camera image](https://github.com/gopro/gpmf-parser/raw/main/docs/readmegfx/CameraIMUOrientationSM.png) to compare what I was getting out through testing, and found that this was the data order. It was the only order that made sense.

I urge you to verify my findings so we'll have better data, and I cannot speak for my past 3 years ago. Best of luck to you, sir!

krishbhk, 2023-06-21:
> @krishbhk Hi! My use case was also specifically MAX only. And my findings were especially for that camera.
>
> I simply tested it out by having a mechanism through gpmf-reader to see and evaluate the [...]

Thanks, I'll try this out myself.