gopro/gpmf-parser #102: potential undefined behavior in OpenMP4Source()

invd, 2020-07-09:
In `OpenMP4Source()`, the following three variables are defined as signed 32 bit integers:

https://github.com/gopro/gpmf-parser/blob/98aff124b6b5d01b42da2d0ffada4ea99758929f/demo/GPMF_mp4reader.c#L350-L352

However, it appears that 
- 32 bit of data is written into them via `fread()` and values are shifted with `<<24` via the `BYTESWAP32` macro
- the represented values are (likely) only supposed to be positive

This can lead to undefined behaviour. Switching the `int32_t` to `uint32_t` might be a solution, but this should be checked by someone that is more familiar with the code.

dnewman-gpsw, 2020-07-10:
From the Quicktime spec, one of the fields can be -1.  From the Quicktime spec "If this field is set to –1, it is an empty edit."  Safer to test of 0xffffffff.

invd, 2020-07-10:
Some debugging information: the `runtime error: left shift of 255 by 24 places cannot be represented in type 'int'` is present in the following three lines:
https://github.com/gopro/gpmf-parser/blob/98aff124b6b5d01b42da2d0ffada4ea99758929f/demo/GPMF_mp4reader.c#L359-L361

In my opinion, value checks would have to be applied before the swapping to prevent the undefined behaviour from being reachable.

Also, note that the offset calculation can potentially overflow:
`runtime error: signed integer overflow: -2097149565 + -402652929 cannot be represented in type 'int'`
https://github.com/gopro/gpmf-parser/blob/98aff124b6b5d01b42da2d0ffada4ea99758929f/demo/GPMF_mp4reader.c#L364

dnewman-gpsw, 2020-08-20:
Good feedback, fixed in the current develop branch

closed 2020-08-20 by dnewman-gpsw