gopro/gpmf-parser #105: data lose between video segments

CanCanZeng, 2020-08-21:
Hi, I find that there are some data lost between video segments, including video data, imu data and gps. For example, the end timestamp of the video is 529.087s, but the timestamp of the start of next video is 531.131s, and I look into the videos, I'm pretty sure that the picture datas are lost. How does this come?

Besides that, the mp4 file exported from .360 file may lost some imu and gps data, because the video lenth is 530s, but imu and gps data just last 510s. This problem happens sometimes, not all of my videos have this problem, but a large part of them come into this problem.

My device is GoPro Max, and I take video in panoramic mode.

CanCanZeng, 2020-08-24:
Hi @dnewman-gpsw, you can download the videos from here:
https://www.dropbox.com/sh/x6skyv87ffyjm1d/AAB99iHH3WOWrR9MrmPBKZnBa?dl=0

> Which tool (and platform) for the .360 files were you exporting with GPMF?
I use GoPro Max Exporter on Windows 10.

dnewman-gpsw, 2020-08-26:
Your camera is running very old, and likely buggy firmware (v1.01).  The camera has had 2 or 3 releases since then (now V1.50.)  Please update you camera and this issue should go away.

closed 2020-08-26 by dnewman-gpsw