gopro/gpmf-parser #107: Is it possible to concatenate videos and imu datas

CanCanZeng, 2020-09-01:
Hi, is it possible to concatenate videos and imu datas? I find that there is no image lose between videos, but I'm not sure if it is so to imu data, since I find that the time lapse between image and imu of first video segment is about 20ms, but the time lapse increse to about 400ms in second video segment. I also find that the imu data in first video lasts less than image data about 400ms, this implys that there maybe no data lose between video segments, and we can cancatenate them. But I'm not sure how to tackle the timestamp problem, since the timestamp of imu is depend on video timestamp.

By the way, I use video timestamp directly from opencv, not extracted from gpmf-parser, but the timestamps of imu are extracted from gpmf-parser. Since I find that the time lapse is more stable if I use timestamp from opencv, and it is not affected by the firmware version.

dnewman-gpsw, 2020-09-01:
There should be no loss in the GPMF data, as no data is flushed between video chapters.  If you concatenate video and metadata into a new file, all data will remain in sync. The GoPro Plus cloud service does this automatically.  I think your measurement of a 400ms sync loss, would either be due to old firmware (some old versions have bugs) or a coding error.  Sync is maintained over chapters, as several tools in our our ecosystem depend on it.

CanCanZeng, 2020-09-02:
Thank you for your apply!
I have upgraded the firmware to the latest 1.5, so it cannot be caused by old version firmware.
And yes, I cancatenate video and metadata to a new file, no data are lost and every thing runs well.
And it's true that there will be time lapse in second chapter because the lenth of video and imu in first chapter are not the same.

dnewman-gpsw, 2020-09-02:
Length of video and metadata are the same in the first chapter, plus or minus a IMU sample.  However the important data like CORI is frame sync'd.

CanCanZeng, 2020-09-02:
can you look into this data?
https://www.dropbox.com/s/d42vpfxcolbu6o6/GS010446.LRV?dl=0

dnewman-gpsw, 2020-09-02:
At 25p there is a bug in the LRV length (bug filed,) however not so in the .360.  You will have to use metadata from the .360 file

 "videolength is 481.5200 seconds, with 481.5200 seconds of metadata",
 "video has 12038 frames at 25.000 fps"
All correctly matches.

CanCanZeng, 2020-09-03:
Thank you for your help! Yes, the length of .360 file is the same as metadata!

closed 2020-09-03 by dnewman-gpsw