gopro/gpmf-parser #108: 360 file changed after firmware upgrade

CanCanZeng, 2020-09-02:
Hi, I upgrade the firmware of my gopro max from 1.0.1 to 1.5, and I find that the .360 file has changed. There are redundant pixels in the picture, but these were not exist in old firmware. I read the article https://gopro.com/en/ch/news/max-tech-specs-stitching-resolution, but cannot find a way to handle the redundant pixel. I also read the article about EAC projection model https://blog.google/products/google-ar-vr/bringing-pixels-front-and-center-vr-video, so I can convert the old .360 file to equirectangular video myself. Can you give some tips about how to remove the redundant pixels? or where I can download the old firmware so I can downgrade my device.

The other problem is that with the "new" 360 file, the GoPro Max Exporter cannot close the video stablization function. I mean, even I de-select all video stablization related options, the exported video is still smoothed. But this problem does not happen to the old version .360 file. Is it a bug?   (I judge the video is smoothed by check wether the rotation between adjcent frames in accordance with the gyro data, only if the two data are accordant I can run visual-inertial odometry）

dnewman-gpsw, 2020-09-02:
You can simply remove the 32-ish pixels of overlap, crop it out, the extra pixels help for doing more advanced exposure blending that the camera couldn't do.  You don't want the old firmware, that was very buggy.  The GoPro Player for Windows is out in beta, it might have the function you need (Facebook group is being used for beta testing https://www.facebook.com/groups/658475918284901.)

CanCanZeng, 2020-09-02:
Hi, thank you for your quick reply.
I tried to just remove the overlap pixel, but it seems that the result is not right. Can you explain it more detailly? For example, 
 I extracted the channal 0 from .360 file, and the left image is from colume 0 to 1376, should I remove colume [656, 688], or [672, 704], or [688, 720]? all of these choice seems not right.

I attch one frame from the channel 0 of .360 file for your convenient.
[image]

CanCanZeng, 2020-09-02:
As you mentioned GoPro Player, I install the latest GoPro Player on my Mac, but the exported video is the same with exported on Windows, they are both smoothed.

dnewman-gpsw, 2020-09-02:
The true EAC image is 4032x2688, six cube faces of 1344x1344.  Each panel of encoded HEVC is 4096x1344, should be processed into 4032x1344 to reconstruct the EAC image (three cube faces of 1344x1344 each.) The two overlap regions are 64 pixels wide, 32 extra pixels from each lens. Source panel has these regions: (656 + 32 + 32 + 656) + (1344) + (656 + 32 + 32 + 656) = 4096 pixels (the braces mark the 3 faces.)  So the center cube face is an unmodified 1344x1344.  All the 656 pixel regions are horizontally scaled up to 688 and the extra 32 pixels dropped or preferably blended.  So you will have 688 + 688 + 1344 + 688 + 688 = 4032.  While not obvious, it is pretty simple to do.  This is what GoPro Player is doing, with a multiband blend, for real-time playback (i.e. the operation is fast.)  
  
GoPro Player bugs or usage questions is outside of my area and gpmf-parser, you should contact GoPro support

CanCanZeng, 2020-09-03:
> All the 656 pixel regions are horizontally scaled up to 688 and the extra 32 pixels dropped or preferably blended. So you will have 688 + 688 + 1344 + 688 + 688 = 4032.
I'm confused by this operation, do you mean scale 656 pixel regions to 672, so I the result image is 672 + 672 + 1344 + 672 + 672 = 4032?  

Below is my code
```
            // the original image with duplicated pixels
            cv::Mat before = frame0.colRange(2720, 4096);

            // method 1
            cv::Mat after1(1344, 1344, frame0.type());

            cv::Mat tmp = before.colRange(0, 656).clone();
            cv::resize(tmp, tmp, cv::Size(688, 1344));
            tmp.colRange(0, 672).copyTo(after1.colRange(0, 672));

            tmp = before.colRange(720, 1344).clone();
            cv::resize(tmp, tmp, cv::Size(688, 1344));
            tmp.colRange(16, 688).copyTo(after1.colRange(672, 1344));

            // method 2
            cv::Mat after2;
            cv::Mat tmp2(1344, 1312, frame0.type());
            before.colRange(0, 656).copyTo(tmp2.colRange(0, 656));
            before.colRange(720, 1376).copyTo(tmp2.colRange(656, 1312));
            cv::resize(tmp2, after2, cv::Size(1344, 1344));
```
the original image is
[image]
the result created according to your explaination(after1 in the code) is like this:
[image]
and the result of my understanding is like this:
[image]
it's not perfect but seems to be right. And it still not as good as the old version firmware created .360 file, so I still miss the old version of firmware, haha.

closed 2020-09-03 by dnewman-gpsw

zjhthu, 2022-08-11:
@CanCanZeng , Hi, have you successfully run VIO with the 360 videos?
I tried to calibrate imu and camera (use the front cube map image), but got weird results. The translation between IMU and the camera is about 1m. 
However, when I use the normal video mode, I was able to get a reasonable translation, e.g., 2cm.

In 360 mode, I find the gyro is not aligned with accl. Gyro is given in 800hz and Accl is given in 200hz. But in norm video mode, both gyro and accl are given in 200hz, and they is no time offset between gyro and accl.

CanCanZeng, 2022-08-12:
@zjhthu I extract pinhole image direct from 360 and interpolated gyro data to accl frequency. And the data can run vins-mono very well.