gopro/gpmf-parser #109: CMake error in install

OgreTransporter, 2020-09-03:
I get the following error message:
```
4>-- Installing: D:/GoPro-gpmf-parser/install/lib/gpmf-parserd.lib
4>CMake Error at cmake_install.cmake:52 (file):
4>  file INSTALL cannot find
4>  "D:/GoPro-gpmf-parser/src/gpmf-parser.pc": File exists.
```

I recommend to change the two lines:

https://github.com/gopro/gpmf-parser/blob/9c07302e1f7e155138dc9f8de7ceb18b30e61a08/CMakeLists.txt#L17

in
```
configure_file("${CMAKE_SOURCE_DIR}/${PROJECT_NAME}.pc.in" "${CMAKE_BINARY_DIR}/${PROJECT_NAME}.pc" @ONLY)
```
and
https://github.com/gopro/gpmf-parser/blob/9c07302e1f7e155138dc9f8de7ceb18b30e61a08/CMakeLists.txt#L21

in
```
install(FILES "${CMAKE_BINARY_DIR}/${PROJECT_NAME}.pc" DESTINATION "lib/pkgconfig")
```

dnewman-gpsw, 2020-09-03:
Okay, I pushed this changes to the develop branch.  What system and CMake version did you see an error?

OgreTransporter, 2020-09-04:
Windows with CMake 3.17.x and CMake 3.18.x

I think the error is the specification `install(FILES "${PROJECT_NAME}.pc" DESTINATION "lib/pkgconfig") `. The PC file does not have an absolute path, so I guess you run CMake in the source directory. Because then the intermediate directory - where the files are written with configure_file - is identical to the source directory. So it does not work if you create the build files in a different directory.

dnewman-gpsw, 2020-09-04:
Thanks. Makes sense.

closed 2020-09-04 by dnewman-gpsw