gopro/gpmf-parser #112: Hero9 GPS accuracy

JuanIrache, 2020-09-21:
Hello @dnewman-gpsw ,

Not sure this is the right place to report on this, as it might be hardware related or anecdotal (a bad unit). Let me know if you want this sent somewhere else.

I noticed my HERO9 was struggling with the GPS data and decided to run some tests side by side with the HERO8. I posted the results here: https://community.gopro.com/t5/GoPro-Telemetry-GPS-overlays/Hero-9-for-telemetry/gpm-p/724365#M635

To sum up, the HERO9 seems to lose any GPS accuracy after some seconds, although it still reports good Fix and Precision values (at least for some data samples).

It might just be my unit, so unless a firmware update that fixes this is expected I will try to get it replaced and test it again.

dnewman-gpsw, 2020-09-25:
We are investigating the GPS precision issues.  I have included your analysis in the bug report.

JuanIrache, 2020-09-25:
Thanks. I talked to other owners that seem to have a good unit, GPS-wise

JuanIrache, 2020-10-05:
Update: After talking to many users, the problem seems very widespread. I could only find a couple of users with good GPS samples. Either they got good hardware and the rest of the units have some issue, or they were lucky when recording their videos (in my experience 3%-8% of video files have close to usable GPS data), which could indicate that this is solvable with a firmware update.

I suppose you can't comment on whether it's a firmware or a hardware issue.

Thank you.

dnewman-gpsw, 2020-10-05:
I'm still following the issue. No idea yet why this is.

djovaniii, 2020-10-20:
Hey,

I'm just pinging for an update.
Facing the same issue, and so far no clew wheter this is going to be fixed or not.
If needed I can provide more data and samples

dnewman-gpsw, 2020-10-20:
We have confirmed the issue and are looking for a good firmware (update) to address it.

xiongjiabin, 2020-10-22:
Looking forward to the new firmware to address this problem.

idaniel-86, 2020-10-22:
Hi,
Joining the conversation as well to hear about the fix. None of my videos report telemetry/gps data even when gos has been always enabled.

dnewman-gpsw, 2020-10-22:
I'm closing this ticket as it is not a bug related to the GPMF-parser.

closed 2020-10-22 by dnewman-gpsw